osrf/subt #388: Message header stamp differs between IMU and camera

TSCarlos, 2020-04-10:
Using the urban_qual branch, and robot configuration X4_5, the camera header stamp lags way behind the IMU header stamp

nkoenig, 2020-04-10:
Can you please provide log files, a reproducible example, and a group id of an instance where this problem occurred on Cloudsim?

rsawtell, 2020-04-22:
I'm having this too in cave_qual and master branches on local dev. Nothing is required to reproduce it other than the run the simulator (don't even need to launch our solution). It appears to be the same problem that popped up before with images lagging behind everything in general.

[image]

Not sure if this is related, but I did notice this error in the simulator info: "[Err] [LevelManager.cc:134] Could not find a plugin tag with name ignition::gazebo. Levels and distributed simulation will not work.". I have ignition-gazebo2 installed, as well as libignition-gazebo2-plugins. Have the package dependencies changed since migrating to git?

rsawtell, 2020-04-22:
On further consideration, it appears to be a problem with the cave_qual branch and not the master branch

nkoenig, 2020-04-22:
I believe you meant to use the `cave_qual_release` branch. This branch has now been merged in `master` in pull request #405 .

nkoenig, 2020-08-17:
I tested using the following:

1. `ign launch -v 4 cave_circuit.ign worldName:=cave_qual robotName1:=X1 robotConfig1:=X1_SENSOR_CONFIG_1`
2. `rostopic echo /X1/front/image_raw | grep -A 3 stamp`
3. `rostopic echo /X1/imu/data | grep -A 3 stamp`

The output from step 2 looks like:
```
stamp: 
    secs: 332
    nsecs: 952000000
  frame_id: "X1/base_link/camera_front"
--
  stamp: 
    secs: 333
    nsecs:         0
  frame_id: "X1/base_link/camera_front"
--
  stamp: 
    secs: 333
    nsecs:  52000000
  frame_id: "X1/base_link/camera_front"
--
  stamp: 
    secs: 333
    nsecs: 100000000
  frame_id: "X1/base_link/camera_front"
--
  stamp: 
    secs: 333
    nsecs: 152000000
  frame_id: "X1/base_link/camera_front
```

And the output from step 3 looks like:

```
  stamp: 
    secs: 333
    nsecs:  60000000
  frame_id: "X1/base_link/imu_sensor"
--
  stamp: 
    secs: 333
    nsecs:  80000000
  frame_id: "X1/base_link/imu_sensor"
--
  stamp: 
    secs: 333
    nsecs: 100000000
  frame_id: "X1/base_link/imu_sensor"
--
  stamp: 
    secs: 333
    nsecs: 120000000
  frame_id: "X1/base_link/imu_sensor"
--
  stamp: 
    secs: 333
    nsecs: 140000000
  frame_id: "X1/base_link/imu_sensor"
```

Can you confirm that you see something different?

nkoenig, 2020-08-24:
Closing since I believe this is fixed.

closed 2020-08-24 by nkoenig