osrf/subt #406: Camera image sensor center (principal point) off by 1.

pauljurczak, 2020-04-21:
It seems that most (all) camera principal point coordinates in `model.sdf` files and consequently in `sensor_msgs/CameraInfo` are off by 1. For example, 640x480 VGA RGBD camera has principal point set to (320.5, 240.5). Assuming the lowest pixel coordinates of (0, 0), principal point (geometric center of image sensor) of this camera is (319.5, 239.5). It is given by `(width/2-0.5, height/2-0.5)` equation not `((width+1)/2, (height+1)/2)` used in `model.sdf` file.

Theoretically, the principal point can be placed anywhere on image sensor plane, but I'm assuming that your intention was to place it centrally to achieve a symmetric pinhole camera model.

acschang, 2021-06-11:
The discrepancy in the principle point is reflected in the `camera_info` topic for each camera stream.  Fixing the reported issue would require a subtle change to the majority of cameras modeled in the repository which could adversely affect team solutions thus this issue cannot be resolved until after Finals.

pauljurczak, 2021-06-11:
No problem. This is very minor, so I'm closing it now.

closed 2021-06-11 by pauljurczak