osrf/subt #408: Incomplete model closure in start area

rsawtell, 2020-04-22:
One of the boundary rocks in the start area doesn't fully reach to the ground, our robot found it and drove into it, getting lodged partly inside a rock.

I apologize for the cruddy pictures but the simulator window is very dark.

Approximate location of the problem (circled in red):
[image]

Visual of the problem area:
[image]

Poor X2 stuck halfway inside a rock, it was only trying to escape the labrynth so it could be free :
[image]

nkoenig, 2020-04-22:
Thanks for point this out. We are working on a fix.

iche033, 2020-04-23:
uploaded new `Cave Starting Area` model onto fuel:
https://app.ignitionrobotics.org/openrobotics/fuel/models/Cave%20Starting%20Area

rsawtell, 2020-04-30:
Not sure if it's related, but I am also seeing these errors running local:

[GUI] [Err] [Material.cc:155] Unable to find texture[../media/textures/CaveWall_Albedo.png] in path[###/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/Cave Starting Area/3//meshes]
[GUI] [Err] [Material.cc:155] Unable to find texture[../media/textures/Grass_Albedo.jpg] in path[###/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/Cave Starting Area/3//meshes]
[GUI] [Err] [Material.cc:155] Unable to find texture[../media/textures/StriatedRock_Albedo.png] in path[###/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/Cave Starting Area/3//meshes]

nkoenig, 2020-05-01:
That  image looks like the user camera's clipping plane has interested the mesh, and clipped part of the tunnel. Can you reproduce that using a camera on a robot?

rsawtell, 2020-05-01:
Yes, here it is from an X1. Notice that the hole shows up in the lidar scan as well.

[image]

iche033, 2020-05-04:
the gap in the cave entrance area should now be fixed too. Model on fuel has been updated:
https://app.ignitionrobotics.org/openrobotics/fuel/models/Cave%20Starting%20Area

nkoenig, 2020-05-07:
New docker images with the patches have been released. I'll close this for now. Please re-open if more holes in the staging area are found.

closed 2020-05-07 by nkoenig