osrf/subt #414: Has someone tried to project laser point on visual image? Please give me a suggestion.

shibowing, 2020-05-07:
Hi! I found a strange thing. Please correct me if I am wrong.

I want to project laser point to the camera frame. Therefore, I want to know the extrinsic of camera and laser. However, I found that I can't project any laser point on camera frame.  I am not sure if I got correct extrinsic and intrinsic matrix.

1. May I ask if every robot can project laser point on camera frame? The prerequisite is that camera and laser should have overlap area.

I will illustrate what I did on projecting laser point. 

Step1: I use this command to run our simimulator.

ign launch -v 4 cave_circuit.ign worldName:=cave_qual robotName1:=X1 robotConfig1:=X1_SENSOR_CONFIG_4

Step2: I got extrinsic matrix from tf tree and intrinsic matrix from topic. The more detail can be found through  attached image.

Ros.Camera topic : "/X1/front/image_raw"
Ros.Velodyne topic : "/X1/points" 

The specific  intrinsic matrix and extrinsic  are as following.

# Camera calibration and distortion parameters (OpenCV) 
Camera.width: 1280
Camera.height: 960

Camera.fx: 1108.5
Camera.fy: 1108.5
Camera.cx: 640.5
Camera.cy: 480.5

# CamLidar Extrinsic Parameter
extrinsicMatrix: !!opencv-matrix
   rows: 3
   cols: 4
   dt: d
   data: [ 1.0, 0.0, 0.0, -0.35,
           0.0, 1.0, 0.0, 0.0,
           0.0,  0.0, 1.0, 0.134]

However, I can't project point. May I ask is there any thing wrong?
[image]

mjcarroll, 2020-05-11:
This is due to a difference in the way that Gazebo and ROS handle their optical coordinate frames.  This is addressed in https://github.com/osrf/subt/pull/413, which adds a new coordinate frame that adheres to [REP103](https://www.ros.org/reps/rep-0103.html#id22).

mjcarroll, 2020-05-18:
Now that #413 is in, this should be fixed.  I'm going to close for now, feel free to reopen if the fix doesn't work for you.

closed 2020-05-18 by mjcarroll