osrf/subt #418: UAV Roll and Pitch Response

npmarion, 2020-05-14:
When using the cmd_vel commands with the UAVs, the velocity commands for roll and pitch (angular y and x) do not  cause any type of motion on the robots.   Would it be possible to add in this capability so that these parameters could be controlled?

pauljurczak, 2020-05-16:
Pitch control would help to overcome rather narrow camera field of view.

pauljurczak, 2020-05-16:
I just realized that dynamics of UAV places constraints on roll and pitch given a fixed trajectory (position in time), so there is no point in controlling them independently.

npmarion, 2020-05-18:
Being able to control the angle of the UAV would be extremely useful, especially for dealing with camera angle, or adjusting where the 2D lidar is able to scan.  Using just the XY velocity commands provides us some tilt, however it is still quite a small angle.  Controlling roll and pitch, even if it induces some type of horizontal/vertical movement would be beneficial to addressing these types of issues.