osrf/subt #420: Version update on Ubuntu 18.04

pauljurczak, 2020-05-16:
Ubuntu Software Updater automagically updates at least some of Ignition components as shown on this screen capture:

[image]

Does it substitute for or overlap with downloading the new release from github and rebuilding `catkin` workspace?

mjcarroll, 2020-05-18:
> Does this update through Software Updater substitute for or overlap with downloading the new release from github and rebuilding catkin workspace?

This is updating the underlying simulation software.  Because ignition releases stay ABI-compatible, it should not be necessary to rebuild your `catkin` workspace when these packages update.  Ignition releases are separate from `subt` releases, but often have fixes or enhancements for the `subt` environment.

> BTW, I just noticed that there are no releases on github. Should we clone the master branch?

For the cave circuit, yes.  There are also corresponding branches for `urban_circuit` and `tunnel_circuit`.

pauljurczak, 2020-05-18:
@mjcarroll  Thank you. I noticed `cave_feature_release1` branch, which is pretty fresh. Should I still favor `master` branch over it?

closed 2020-05-19 by pauljurczak