osrf/subt #421: No TF transform between laser and imu, laser and camera

shibowing, 2020-05-17:
Hi, 

I just merge my code with our latest one. I found that our latest code cannot obtain the tf information between laser and imu, laser and camera. However, in previous version, the tf between sensors is published. 

The more information can be found here.

commit 0eed307c9b25bc0cfdf3a1577b090fa64926c384 (HEAD -> master, origin/master, origin/HEAD)
Author: iche033 <ichen@osrfoundation.org>
Date:   Thu May 14 13:54:17 2020 -0700

    Add optical frame publisher (#413)
    
    * check topic heartbeat and log /X*/pose topics
    
    * use pose_v and subt to pose_static
    
    * cleanup
    
    * Rename parameters
    
    * move heartbeat check into its own thread
    
    * add optical frame publisher and update vehicle topics
    
    Signed-off-by: Ian Chen <ichen@osrfoundation.org>
    
    * fix rotation
    
    Signed-off-by: Ian Chen <ichen@osrfoundation.org>
    
    * update optical frame publisher args
    
    Signed-off-by: Ian Chen <ichen@osrfoundation.org>
    
    * lazy subscription
    
    Signed-off-by: Ian Chen <ichen@osrfoundation.org>
    
    Co-authored-by: Nate Koenig <nate@openrobotics.org>

commit e615b3160a71ab6a0f00cfbec109830f36125cc8
Author: iche033 <ichen@osrfoundation.org>
Date:   Thu May 14 13:46:27 2020 -0700

    Reduce /<robot_name>/pose topic bandwidth (#402)
    
    * check topic heartbeat and log /X*/pose topics
    
    * use pose_v and subt to pose_static
    
    * cleanup
    
    * Rename parameters
    
    * move heartbeat check into its own thread

commit 9132c0c564f6cf1f26b99167deed2078657e1249
Merge: 4510ab2 da4ba98
Author: Nate Koenig <nkoenig@users.noreply.github.com>
Date:   Wed Apr 22 14:43:47 2020 -0700

    Merge pull request #405 from osrf/cave_qual_release
    
    Cave Qual release

mjcarroll, 2020-05-18:
@shibowing Which robot platform are you using?

shibowing, 2020-05-18:
Thanks for the quick reply!  This is my command:
ign launch -v 4 cave_circuit.ign worldName:=cave_qual robotName1:=X1 robotConfig1:=X1_SENSOR_CONFIG_4

iche033, 2020-05-18:
> I just merge my code with our latest one

Are you compiling from source? If so, can you check to make sure you have `ros-melodic-ros-ign*` packages version 0.9.3, and `libignition-gazebo2-dev` version 2.17.0 ?

nkoenig, 2020-07-17:
Closing due to lack of response.

closed 2020-07-17 by nkoenig