osrf/subt #432: AWS CLI version 2.0 does not work with subt_docker.bash

wolfgangschwab, 2020-05-25:
We set up a new pc and also tried to set up the upload functionality using AWS CLI as described in [Cloudsim Submission](https://github.com/osrf/subt/wiki/Cloudsim%20Submission#instructions).

It seems that the  `subt_docker.bash` script does not work with the current Version 2.0 of AWS CLI . 

1. Using the script we get an error that the command get-login is not available.
2. Using the eval command  eval `aws ecr get-login --no-include-email` results in the same error.

We learned that get-login is replaced in version 2.0 by get-login-password. unfortunately changing this did not solve the issue completely. 
Should we try to install version 1 of AWS CLI? Or will the script be updated to version 2.0?

nkoenig, 2020-05-28:
Thanks for the info. I am running version 1 of the tool. I'll try out version 2 and figure out what needs to be changed.

nkoenig, 2020-05-28:
Can you try the following:
```
aws ecr get-login-password --region us-east-1 | docker login --username AWS --password-stdin 200670743174.dkr.ecr.us-east-1.amazonaws.com
```

nkoenig, 2020-05-28:
Also checkout #435 .

wolfgangschwab, 2020-05-29:
Ok, thanks. Will do as soon as the other computer is available.

georgschwab, 2020-05-29:
The result of the command is:
`
WARNING! Your password will be stored unencrypted in /home/georg/.docker/config.json.
Configure a credential helper to remove this warning. See
https://docs.docker.com/engine/reference/commandline/login/#credentials-store

Login Succeeded
`
so looks good for me.

#435 does not work:
`Logging into docker
WARNING! Your password will be stored unencrypted in /home/georg/.docker/config.json.
Configure a credential helper to remove this warning. See
https://docs.docker.com/engine/reference/commandline/login/#credentials-store

./src/subt/subt_ign/scripts/subt_docker.bash: Zeile 35: Login: Befehl nicht gefunden
Failed to log into docker. Check your AWS credentials.
`

nkoenig, 2020-06-11:
I believe `Befehl nicht gefunden` means "command not found". Can you try each part separately:

I'm assuming the script couldn't find `docker login`, but that seems strange if you can run:
```
aws ecr get-login-password --region us-east-1 | docker login --username AWS --password-stdin 200670743174.dkr.ecr.us-east-1.amazonaws.com
```

georgschwab, 2020-06-13:
Thanks for your feedback. Yes, it means "command not found".

We were able to narrow the problem down to the "$" in front of the command in the script. Removing it resolves the problem for us. Don't know why though.

nkoenig, 2020-06-15:
Does this work, as a replacement to `$(aws ecr get-login-password --region us-east-1 | docker login --username AWS --password-stdin 200670743174.dkr.ecr.us-east-1.amazonaws.com)`:

```
eval 'aws ecr get-login-password --region us-east-1 | docker login --username AWS --password-stdin 200670743174.dkr.ecr.us-east-1.amazonaws.com'
```

georgschwab, 2020-06-15:
Yes. The replacement works.

nkoenig, 2020-08-17:
See pull request #544

nkoenig, 2020-08-20:
PR #544 has been merged.

closed 2020-08-20 by nkoenig