osrf/subt #433: Dockerfile for subt does not work for newly installed system

wolfgangschwab, 2020-05-25:
We set up a new pc. We run the Docker setup and also came to the part where the subt-Docker image is build as described in [Build-the-subt-image](https://github.com/osrf/subt/wiki/Docker%20Install#build-the-subt-image). 
The two `COPY` statements does not work because they are looking for files that are not available. But the files are already downloaded into the image. So the command can be changed to `RUN cp ...`

Changes in detail:

1. old: 'COPY download_models.sh ./'
new: 'RUN cp subt_ws/src/subt/docker/download_models.sh ./'
2. old: 'COPY subt_sim_entry/run_sim.bash ./'
new: 'RUN cp src/subt/docker/subt_sim_entry/run_sim.bash ./'

I think it would be good to change this in the Dockerfile.

nkoenig, 2020-05-28:
You have linked to a deprecated version of the docker install instructions. Was that a mistake?

wolfgangschwab, 2020-05-29:
I'm not aware that this instruction is deprecated. I came there using the route: 
`wiki -> Get Started -> Docker image installation -> Prerequisites: Install Docker`
Where can I find the correct docker install instructions?

nkoenig, 2020-08-17:
See the bold `warning` directly below the `Build the SubT image` header. The top section of the page is fine to follow.

I'll remove the section you are referring to since it's been over a year since SubT transitioned from Gazebo9 to Ignition.

closed 2020-08-17 by nkoenig