osrf/subt #436: Add variable TEAMBASE name

dan-riley, 2020-05-28:
Please make the TEAMBASE name variable.  We already had an infrastructure setup with a different base station name and it was far easier to just change the name in the launch files so we didn't have to change our other code.  It looks like it only needs about 3 or 4 lines of code in each launch file to make it variable.  I was able to hardcode our name by changing lines 1278, 1279 and 1370.  I could probably make an edit and a pull request to make them variable, but probably easier for someone on your end to do it!  This will make it easier next time you do an update so we don't have to update all of those files again.

To clarify, the biggest issue for us was in the communications nodes, which we had some code setup based on an expected static base station.  I wouldn't be surprised if others had a similar setup, so changing it here makes it easier on everyone.

zwn, 2020-05-29:
Let me second that request.

The name of the robot is the only way how to give it any information. We build a single image and by varying the name of the robot we can test different strategies. For one, I'd like to be able to tell teambase how many robots are expected to start. That way the teambase can coordinate the start of the robots, giving us a partial solution to #412.

Another nice feature would involve allowing more characters in the robot name (like a hypen or underscore or dot).

nkoenig, 2020-06-02:
So, the goal is to encode data in the name of the teambase?

zwn, 2020-06-02:
> So, the goal is to encode data in the name of the teambase?

If that question is for me - yes. The expectation was that teambase would be just another robot, thus the fact that it is teambase would be identified just by the config value (so just the part `robotConfigN:=TEAMBASE`) and we would be able to assign name freely.

dan-riley, 2020-06-02:
Not for me, I just want to be able to use whatever name we would like, just like the robots.  But it does sound like @zwn is using characters in the name for other purposes.

nkoenig, 2020-07-08:
The name for a teambase can now be specified both on the command line and through the portal.

closed 2020-07-08 by nkoenig