osrf/subt #438: catkin installtion does not show correct pictures in GUI

wolfgangschwab, 2020-05-29:
We set up a new computer also with the catkin workspace option. After the installation we see again the funny/broken pictures of the tunnel worlds. The same problem has been noticed before on an old computer as reported in issue #198.

[image]
This was produced using the command:
`ign launch -v 4 tunnel_circuit_practice.ign worldName:=tunnel_circuit_practice_01 robotName1:=X1 robotConfig1:=X1_SENSOR_CONFIG_1`

When using urban or cave worlds (`ign launch -v 4 cave_circuit.ign worldName:=simple_cave_01 robotName1:=X2 robotConfig1:=X2_SENSOR_CONFIG_7`) the view is almost completely grey:
[image]

Does anybody have a hint how to solve this issue?

PS: I know that the development can be done in a docker container, but in some cases we use the catkin workspace to easier investigate problems.

pauljurczak, 2020-06-01:
I'm developing in catkin workspace on 2 different PCs and cave worlds work just fine on both of them.

wolfgangschwab, 2020-06-02:
@pauljurczak,  
thanks for the feedback. I'm using a new computer now. Here the problem occurs. At the old computer the development with catkin workspace works. So in principle its OK, but not with my new computer. And I do no know where to look to fix this problem. 

If anybody has an idea. I would appreciate any hint.

zwn, 2020-06-02:
@wolfgangschwab catkin workspace has got more knobs that can be turned wrong. The general suggestions would sound like canned responses but I am not sure if there is anything better:

* setup the upstream apt repos correctly
* make sure nvidia based opengl is working correctly
* update often

However if the simulation works when launched from the docker image then you can try to find out what is different in your host computer setup vs the simulation docker.

Does the simulation docker work correctly on the same computer?

pauljurczak, 2020-06-03:
@wolfgangschwab 
I would also upgrade gcc and g++ to v8.4 (see https://github.com/osrf/subt/issues/444)

wolfgangschwab, 2020-06-04:
Thank you all for the feedback and ideas.
 
@nkoenig: Yes, the docker images work fine. Very confusing.  I also tried to reproduce the catkin workspace I did an a new docker. But there also the graphics look well. 
I have a RTX 2060 super graphics card and I installed the driver using `sudo apt install nvidia-driver-440`. 
Might there be issues with this version? But in the docker `nvidia-smi` also shows driver version 440.59

@pauljurczak : In catkin workspace I have gcc/g++ version 7.5. But in the docker we are using the same version.  

@zwn : opengl sounds for me as a possible problem. I'll check this.

pauljurczak, 2020-06-05:
@wolfgangschwab 
I'm using Nvidia driver 440.59 without problems for catkin build.

nkoenig, 2020-06-05:
* Do you see any red highlighted text in the terminal where you launch simulation?

* Could you try an Urban and/or Cave environment? I'm curious to see if the problem is world dependent.

wolfgangschwab, 2020-06-05:
- There is no red highlighted text in the terminal. 

- The second picture above "shows" the cave world. Only 3 cylinders are visible. It seems that they are coming from the robot. The position fits to the robot and using another config (X1-Config1) shows no cylinders at all.
- I tried to reproduce the issue in my docker by removing 2 packages (opengl and glvnd) which are not installed on the computer. But the issue does not ocure in docker.

wolfgangschwab, 2020-06-05:
Thanks for your support. 
1. Yes I'm running Ubuntu 18.04
2. Yes, I'm using ROS melodic and Ignition Blueprint in the catkin workspace installation, installed using the catkin workspace installation instructions
3. `glxinfo` was not available, after installation of `mesa-utils` I got:
name of display: :1
display: :1  screen: 0
direct rendering: Yes
server glx vendor string: NVIDIA Corporation
server glx version string: 1.4
server glx extensions:

The models have already been downloaded about a week ago, which really took a while. I started the script to download again. 

I added three cylinders in an own small cave-world. The cave part is not shown. But the cylinders are shown:
[image]
The small parts to the right of the first cylinder seem to come from the robot.

pauljurczak, 2020-06-06:
@wolfgangschwab 
I would check that your Nvidia driver is actually running instead of open source one. I'm not sure what is the best way, but try `nvidia-smi` from the terminal or `Nvidia X Server Settings` from the program launcher. There are known issues with proprietary Nvidia driver one has to work around when configuring a new system.

wolfgangschwab, 2020-06-06:
@nate: We have a new day and the `download_models` script has finished. Unfortunately the issue is still there. 
Sounds good that you will be able to test with a RTX graphics card. But I'm affraid that the problem is not related with this nvidia card because I saw this problem last year with an oder PC which does not have an RTX card.  

@pauljurczak : I installed the Nvidia driver 440 using 'sudo apt install nvidia-driver-440'. 'nvidia-smi' gives:

> Sat Jun  6 11:04:49 2020       
> +-----------------------------------------------------------------------------+
> | NVIDIA-SMI 440.59       Driver Version: 440.59       CUDA Version: 10.2     |
> |-------------------------------+----------------------+----------------------+
> | GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
> | Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
> |===============================+======================+======================|
> |   0  GeForce RTX 206...  Off  | 00000000:02:00.0  On |                  N/A |
> | 29%   41C    P8    13W / 175W |    756MiB /  7977MiB |     23%      Default |
> +-------------------------------+----------------------+----------------------+
>                                                                                
> +-----------------------------------------------------------------------------+
> | Processes:                                                       GPU Memory |
> |  GPU       PID   Type   Process name                             Usage      |
> |=============================================================================|
> |    0      2478      G   /usr/lib/xorg/Xorg                            38MiB |
> |    0      2514      G   /usr/bin/gnome-shell                          57MiB |
> |    0      2992      G   /usr/lib/xorg/Xorg                           328MiB |
> |    0      3130      G   /usr/bin/gnome-shell                         282MiB |
> |    0      4894      G   /usr/lib/firefox/firefox                       5MiB |
> |    0      4937      G   /usr/lib/firefox/firefox                       5MiB |
> |    0      5243      G   /usr/lib/firefox/firefox                       5MiB |
> |    0      5284      G   /usr/lib/firefox/firefox                      21MiB |
> |    0      7394      G   /usr/lib/firefox/firefox                       5MiB |
> +-----------------------------------------------------------------------------+
So I would guess that the nvidia driver is used.

wolfgangschwab, 2020-06-06:
I also looked at the illumination of three cylinders I added in my small world. In the docker verison the second cylinder has some shadow from the rocks in front of it:
[image]
In the catkin workspace version the rocks are not taking away any light. But the collision works. The third cylinder is places on top of a rock and does not fall down:
[image]

wolfgangschwab, 2020-06-07:
OK, I found the cause of the problem and a workaround to solve it. The solution is almost unbelievable for me. 

How did I find the solution? The scattered/broken parts of the tunnel worlds are all in vertical or horizontal direction and the length seem to be always rounded to meters. So it could have something to do with rounding or converting of float to int. After many tries I also changed the region settings. I set up my Ubuntu with German  setting. We use in Germany the comma as the decimal point (not the dot as in USA). Changing all regional setting to United States solved the issue. Finally changing the formats in the settings from `Germany` to `United States` is the key. 

Screenshot with Germany formats:
[image]

And the version with the United States formats:
[image]

So I will use the United States settings from now on :-). I spend too much time with this issue. 

Shall we leave the issue open to investigate the reason in the code or shall I close it?

pauljurczak, 2020-06-08:
@wolfgangschwab 
Wow, that was unexpected! Good thing that the numbers a written with digits not words. A Billion in German means something different from a billion in English.

nkoenig, 2020-06-08:
Thanks for posting about your solution. Let's leave this issue open for visibility and as a reminder to us to find a better solution than changing regions.

peci1, 2020-07-28:
God damn this issue :) Could it be caused by something like https://github.com/ros/urdfdom_headers/issues/41 ?

You don't have to change your system locale, it's sufficient to run the launch command with an english locale like this:

    LC_ALL=C LANG=C ign launch -v 4 tunnel_circuit_practice.ign worldName:=tunnel_circuit_practice_01 robotName1:=X1 robotConfig1:=X1_SENSOR_CONFIG_1

I'm not familiar with the .ign launch files, but if they were similar to ROS launch files, these variables could be set in them. Would that be a solution?

nkoenig, 2020-07-28:
This problem was fixed in [SDF](https://osrf-migration.github.io/sdformat-gh-pages/#!/osrf/sdformat/pull-requests/492/page/1). I think we need to go find all the places that interpret strings as floats. @iche033 , can you see where we need to apply a locale fix similar to [this](https://osrf-migration.github.io/sdformat-gh-pages/#!/osrf/sdformat/pull-requests/492/page/1)?

peci1, 2020-07-29:
Any ETA on when will it be released in Melodic?

nkoenig, 2020-07-29:
No ETA because we don't know exactly where all of the problems are.