osrf/subt #444: Issues with installation of the current version

pauljurczak, 2020-06-03:
I had to reinstall my Ubuntu 18.04 and I encountered a few problems. The first one, I suspect is caused by Ubuntu 18.04 version of gcc being 7.5. I had this crash when I launched a test:
```
paul@desktop:~$ ign launch -v 4 cave_circuit.ign worldName:=cave_qual robotName1:=X1 robotConfig1:=X1_SENSOR_CONFIG_1
.....
[Msg] Loading plugin [ignition-rendering2-ogre2]
[Dbg] [EntityComponentManager.cc:627] Using components of type [4425034272867445569] / [ign_gazebo_components.Performer].
[Dbg] [EntityComponentManager.cc:627] Using components of type [11311330372155309806] / [ign_gazebo_components.PerformerLevels].
[Dbg] [SimulationRunner.cc:431] Exiting postupdate worker thread (2)

[Dbg] [SimulationRunner.cc:431] Exiting postupdate worker thread (0)

[Dbg] [SimulationRunner.cc:431] Exiting postupdate worker thread (6
[Dbg] [SimulationRunner.cc:431] [Dbg] [SimulationRunner.cc:431] Exiting postupdate worker thread ()[Dbg] [SimulationRunner.cc:431] [Dbg] [SimulationRunner.cc:431] )[Dbg] [SimulationRunner.cc:431] Exiting postupdate worker thread (15)Exiting postupdate worker thread ()[Dbg] [SimulationRunner.cc:431] 
3))
)
Exiting postupdate worker thread (4)

[Dbg] [SimulationRunner.cc:402] Creating PostUpdate worker threads: 12
[Dbg] [SimulationRunner.cc:415] Creating postupdate worker thread (0)
[Dbg] [SimulationRunner.cc:415] Creating postupdate worker thread (1)
[Dbg] [SimulationRunner.cc:415] Creating postupdate worker thread (2)
[Dbg] [SimulationRunner.cc:415] Creating postupdate worker thread (3)
[Dbg] [SimulationRunner.cc:415] Creating postupdate worker thread (4)
[Dbg] [SimulationRunner.cc:415] Creating postupdate worker thread (5)
[Dbg] [SimulationRunner.cc:415] Creating postupdate worker thread (6)
[Dbg] [SimulationRunner.cc:415] Creating postupdate worker thread (7)
[Dbg] [SimulationRunner.cc:415] Creating postupdate worker thread (8)
[Dbg] [SimulationRunner.cc:415] Creating postupdate worker thread (9)
[Dbg] [SimulationRunner.cc:415] Creating postupdate worker thread (10)
[Dbg] [EntityComponentManager.cc:627] Using components of type [14423792621074650228] / [ign_gazebo_components.JointVelocityCmd].
[Dbg] [EntityComponentManager.cc:627] Using components of type [9436740692853731264] / [ign_gazebo_components.JointPosition].
[Dbg] [Sensors.cc:344] Initialization needed
Stack trace (most recent call last) in thread 32106:
#18   Object "", at 0xffffffffffffffff, in 
#17   Source "/build/glibc-OTsEL5/glibc-2.27/misc/../sysdeps/unix/sysv/linux/x86_64/clone.S", line 95, in  [0x7febf453d88e]
#16   Source "/build/glibc-OTsEL5/glibc-2.27/nptl/pthread_create.c", line 463, in start_thread [0x7febf42046da]
#15   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7febf0add6de, in std::error_code::default_error_condition() const
#14   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-2/plugins/libignition-gazebo-sensors-system.so", at 0x7febbcedc4a6, in ignition::gazebo::v2::systems::SensorsPrivate::RenderThread()
#13   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-2/plugins/libignition-gazebo-sensors-system.so", at 0x7febbcedaa10, in ignition::gazebo::v2::systems::SensorsPrivate::WaitForInit()
#12   Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo2-rendering.so.2", at 0x7febbcc7668c, in ignition::gazebo::v2::RenderUtil::Init()
#11   Object "/usr/lib/x86_64-linux-gnu/libignition-rendering2.so.2", at 0x7febca3b2057, in ignition::rendering::v2::RenderEngineManager::Engine(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#10   Object "/usr/lib/x86_64-linux-gnu/libignition-rendering2.so.2", at 0x7febca3b1e42, in ignition::rendering::v2::RenderEngineManagerPrivate::Engine(std::_Rb_tree_iterator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, ignition::rendering::v2::RenderEngine*> >, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#9    Object "/usr/lib/x86_64-linux-gnu/libignition-rendering2.so.2", at 0x7febca3bc39d, in ignition::rendering::v2::BaseRenderEngine::Load(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)
#8    Object "/usr/lib/x86_64-linux-gnu/libignition-rendering2-ogre2.so", at 0x7feb8084faca, in ignition::rendering::v2::Ogre2RenderEngine::LoadImpl(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)
#7    Object "/usr/lib/x86_64-linux-gnu/libignition-rendering2-ogre2.so", at 0x7feb8084f714, in ignition::rendering::v2::Ogre2RenderEngine::LoadAttempt()
#6    Object "/usr/lib/x86_64-linux-gnu/libignition-rendering2-ogre2.so", at 0x7feb8084b16b, in ignition::rendering::v2::Ogre2RenderEngine::LoadPlugins()
#5    Object "/usr/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7feb73b4a312, in Ogre::Root::loadPlugin(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#4    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/RenderSystem_GL3Plus.so", at 0x7feb68be9443, in dllStartPlugin
#3    Object "/usr/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7feb73b4a0e5, in Ogre::Root::installPlugin(Ogre::Plugin*)
#2    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/RenderSystem_GL3Plus.so", at 0x7feb68c00c4d, in Ogre::GL3PlusPixelFormatToShaderType::getPixelFormatType(Ogre::PixelFormat) const
#1    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/RenderSystem_GL3Plus.so", at 0x7feb68c07360, in Ogre::GL3PlusRenderSystem::GL3PlusRenderSystem()
#0    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/RenderSystem_GL3Plus.so", at 0x7feb68bde115, in Ogre::GLSLShaderManager::unregisterShaderFactory(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
Segmentation fault (Invalid permissions for mapped object [0x7febb85ac0d8])
Segmentation fault (core dumped)
```

I upgraded gcc and g++ to v8.4 and test is running fine but I see this error message:
```
paul@desktop:~$ ign launch -v 5 cave_circuit.ign worldName:=cave_qual robotName1:=X1 robotConfig1:=X1_SENSOR_CONFIG_1
.....
========

PARAMETERS
 * /X1/robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.5

NODES

ROS_MASTER_URI=http://localhost:11311

No processes to monitor
shutting down processing monitor...
... shutting down processing monitor complete
[Dbg] [EntityComponentManager.cc:627] Using components of type [2251689575469537287] / [ign_gazebo_components.World].
[Dbg] [EntityComponentManager.cc:627] Using components of type [13994732549916512682] / [ign_gazebo_components.Name].
[Dbg] [EntityComponentManager.cc:627] Using components of type [12592746352568925681] / [ign_gazebo_components.Gravity].
[... cut ...]
[Dbg] [EntityComponentManager.cc:627] Using components of type [15943768124495574352] / [ign_gazebo_components.WorldLinearVelocitySeed].
[Dbg] [EntityComponentManager.cc:627] Using components of type [3297509811873971798] / [ign_gazebo_components.ParentEntity].
[Dbg] [EntityComponentManager.cc:627] Using components of type [17100615127981600159] / [ign_gazebo_components.Scene].
[Err] [LevelManager.cc:139] Could not find a plugin tag with name ignition::gazebo. Levels and distributed simulation will not work.
```
Perhaps the installation script should be updated to install `gcc 8` and `g++ 8`?

nkoenig, 2020-06-09:
What version of ign-gazebo do you have installed, and are you on the master branch of this repo?

pauljurczak, 2020-06-09:
Yes, it was the master branch.

pauljurczak, 2020-07-04:
BTW, I noticed the same error with the newest release. As far as I can tell, simulation works normally.
```
... shutting down processing monitor complete
[Dbg] [EntityComponentManager.cc:627] Using components of type [2251689575469537287] / [ign_gazebo_components.World].
[Dbg] [EntityComponentManager.cc:627] Using components of type [13994732549916512682] / [ign_gazebo_components.Name].
[Dbg] [EntityComponentManager.cc:627] Using components of type [12592746352568925681] / [ign_gazebo_components.Gravity].
[Dbg] [EntityComponentManager.cc:627] Using components of type [13224937992534617849] / [ign_gazebo_components.MagneticField].
[Dbg] [EntityComponentManager.cc:627] Using components of type [3630648173860223239] / [ign_gazebo_components.PhysicsEnginePlugin].
[Dbg] [EntityComponentManager.cc:627] Using components of type [8753193699724811771] / [ign_gazebo_components.Wind].
[Dbg] [EntityComponentManager.cc:627] Using components of type [12173050716021724529] / [ign_gazebo_components.WorldLinearVelocity].
[Dbg] [EntityComponentManager.cc:627] Using components of type [15943768124495574352] / [ign_gazebo_components.WorldLinearVelocitySeed].
[Dbg] [EntityComponentManager.cc:627] Using components of type [3297509811873971798] / [ign_gazebo_components.ParentEntity].
[Dbg] [EntityComponentManager.cc:627] Using components of type [17100615127981600159] / [ign_gazebo_components.Scene].
[Err] [LevelManager.cc:139] Could not find a plugin tag with name ignition::gazebo. Levels and distributed simulation will not work.
```

nkoenig, 2020-08-17:
That last error
```
[Err] [LevelManager.cc:139] Could not find a plugin tag with name ignition::gazebo. Levels and distributed simulation will not work.
```

Was benign and has been fixed in pull request #550.

I wasn't able to reproduce the original error. I have gcc-7.5.0 installed on Ubuntu 18.04. I've followed the [Catkin system setup](https://github.com/osrf/subt/wiki/Catkin%20System%20Setup), and can run `ign launch -v 4 cave_circuit.ign worldName:=cave_qual robotName1:=X1 robotConfig1:=X1_SENSOR_CONFIG_1` without problems. Do you have the same setup?

pauljurczak, 2020-08-17:
My current status is: 2020-08-11 Docker images are running fine, but I still see this error message.
```
paul@desktop:~/subt_testbed$ ./run.bash osrf/subt-virtual-testbed cave_circuit.ign worldName:=cave_qual robotName1:=X1 robotConfig1:=EXPLORER_DS1_SENSOR_CONFIG_1 headless:=false
......
ROS_MASTER_URI=http://localhost:11311

No such file or directory: /home/developer/subt_ws/install/share/explorer_ds1_sensor_config_1/urdf/model.xacro
process[bridge_logger-4]: started with pid [135]
RLException: Invalid <param> tag: Cannot load command parameter [X1/robot_description]: command [['/opt/ros/melodic/lib/xacro/xacro', '/home/developer/subt_ws/install/share/explorer_ds1_sensor_config_1/urdf/model.xacro', 'name:=X1']] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro '$(find explorer_ds1_sensor_config_1)/urdf/model.xacro' name:=$(arg name)" name="$(arg name)/robot_description"/>
The traceback for the exception was written to the log file
process[rostopic_stats_logger-5]: started with pid [159]
.......
[Err] [LevelManager.cc:139] Could not find a plugin tag with name ignition::gazebo. Levels and distributed simulation will not work.
```
The same with Docker Compose running Cloudsim:
```
sim_1        | [Err] [LevelManager.cc:139] Could not find a plugin tag with name ignition::gazebo. Levels and distributed simulation will not work.
```
It doesn't seem to affect functionality, so it is a low priority issue for me.

pauljurczak, 2020-08-17:
2020-08-11 version of Catkin build from master branch fails:
```
[ 77%] Building CXX object subt/subt_ign/CMakeFiles/validate_connections.dir/src/ConnectionValidatorPrivate.cc.o
In file included from /usr/include/yaml-cpp/node/iterator.h:13,
                 from /usr/include/yaml-cpp/node/impl.h:11,
                 from /usr/include/yaml-cpp/yaml.h:17,
                 from /home/paul/subt_ws/src/subt/subt_ign/src/GameLogicPlugin.cc:18:
/usr/include/yaml-cpp/node/detail/iterator.h: In member function ‘void YAML::detail::iterator_base<V>::increment()’:
/usr/include/yaml-cpp/node/detail/iterator.h:48:54: error: ‘next’ is not a member of ‘boost’
   void increment() { this->base_reference() = boost::next(this->base()); }
                                                      ^~~~
/usr/include/yaml-cpp/node/detail/iterator.h:48:54: note: suggested alternatives:
In file included from /usr/include/c++/8/bits/stl_algobase.h:66,
                 from /usr/include/c++/8/bits/char_traits.h:39,
                 from /usr/include/c++/8/ios:40,
                 from /usr/include/yaml-cpp/parser.h:10,
                 from /usr/include/yaml-cpp/yaml.h:10,
                 from /home/paul/subt_ws/src/subt/subt_ign/src/GameLogicPlugin.cc:18:
/usr/include/c++/8/bits/stl_iterator_base_funcs.h:213:5: note:   ‘std::next’
     next(_InputIterator __x, typename
     ^~~~
In file included from /usr/include/boost/mpl/next.hpp:17,
                 from /usr/include/boost/mpl/bind.hpp:25,
                 from /usr/include/boost/mpl/lambda.hpp:18,
                 from /usr/include/boost/mpl/apply.hpp:25,
                 from /usr/include/boost/iterator/iterator_facade.hpp:36,
                 from /usr/include/yaml-cpp/node/detail/node_iterator.h:12,
                 from /usr/include/yaml-cpp/node/detail/iterator.h:12,
                 from /usr/include/yaml-cpp/node/iterator.h:13,
                 from /usr/include/yaml-cpp/node/impl.h:11,
                 from /usr/include/yaml-cpp/yaml.h:17,
                 from /home/paul/subt_ws/src/subt/subt_ign/src/GameLogicPlugin.cc:18:
/usr/include/boost/mpl/next_prior.hpp:29:8: note:   ‘boost::mpl::next’
 struct next
        ^~~~
[ 78%] Building CXX object subt/subt_ign/CMakeFiles/ConnectionValidator.dir/src/ConnectionHelper.cc.o
[ 79%] Building CXX object subt/subt_ign/CMakeFiles/CommsBrokerPlugin.dir/src/VisibilityTable.cc.o
[ 79%] Building CXX object subt/subt_ign/CMakeFiles/ConnectionValidator.dir/src/SimpleDOTParser.cc.o
[ 79%] Building CXX object subt/subt_ign/CMakeFiles/validate_connections.dir/src/ConnectionHelper.cc.o
[ 80%] Linking CXX shared library /home/paul/subt_ws/devel/lib/libConnectionValidator.so
[ 80%] Building CXX object subt/subt_ign/CMakeFiles/CommsBrokerPlugin.dir/src/SimpleDOTParser.cc.o
[ 80%] Built target ConnectionValidator
Scanning dependencies of target VisibilityPlugin
[ 81%] Building CXX object subt/subt_ign/CMakeFiles/VisibilityPlugin.dir/src/VisibilityPlugin.cc.o
subt/subt_ign/CMakeFiles/GameLogicPlugin.dir/build.make:62: recipe for target 'subt/subt_ign/CMakeFiles/GameLogicPlugin.dir/src/GameLogicPlugin.cc.o' failed
make[2]: *** [subt/subt_ign/CMakeFiles/GameLogicPlugin.dir/src/GameLogicPlugin.cc.o] Error 1
CMakeFiles/Makefile2:4507: recipe for target 'subt/subt_ign/CMakeFiles/GameLogicPlugin.dir/all' failed
make[1]: *** [subt/subt_ign/CMakeFiles/GameLogicPlugin.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 82%] Building CXX object subt/subt_ign/CMakeFiles/VisibilityPlugin.dir/src/VisibilityTable.cc.o
[ 84%] Linking CXX shared library /home/paul/subt_ws/devel/lib/libCommsBrokerPlugin.so
[ 85%] Building CXX object subt/subt_ign/CMakeFiles/validate_connections.dir/src/SimpleDOTParser.cc.o
[ 85%] Built target CommsBrokerPlugin
[ 85%] Building CXX object subt/subt_ign/CMakeFiles/VisibilityPlugin.dir/src/SimpleDOTParser.cc.o
[ 85%] Linking CXX executable /home/paul/subt_ws/devel/lib/subt_ign/validate_connections
[ 85%] Built target validate_connections
[ 86%] Linking CXX shared library /home/paul/subt_ws/devel/lib/libVisibilityPlugin.so
[ 86%] Built target VisibilityPlugin
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make install -j4 -l4" failed
paul@extra:~/subt_ws$ g++ --version
g++ (Ubuntu 8.4.0-1ubuntu1~18.04) 8.4.0
```
I'm using g++ 8.4 because CUDA doesn't support g++ 7.5 any longer.

nkoenig, 2020-08-18:
I'm not entirely sure why your catkin build would fail in the yaml library. You could try making sure that you've followed all the steps in [catkin install](https://github.com/osrf/subt/wiki/Catkin%20System%20Setup). I would suspect that there is a bad install of some library on your system.

pauljurczak, 2020-08-19:
It builds fine on another PC. The error was most likely caused by boost 1.72 installed alongside of boost 1.65.

closed 2020-08-19 by pauljurczak