osrf/subt #452: Linux dependency hell

pauljurczak, 2020-06-08:
I needed `npm` package for my development environment on Ubuntu 18.04.4, which in turn needs `nodejs-dev node-gyp libssl1.0-dev`. Package manager suggests removal of Igintion and some ROS infrastructure, which I unfortunately agreed to without reading the fine print. Here are the gory details:
```
paul@extra:~$ sudo apt-get install nodejs-dev node-gyp libssl1.0-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  fonts-lato freeglut3 gdal-data hddtemp ignition-gazebo2 ignition-tools libaec0 libarmadillo8 libarpack2 libassimp-dev
  libassimp4 libassuan-dev libavcodec-dev libavdevice-dev libavdevice57 libavfilter-dev libavformat-dev libavresample-dev
  libavutil-dev libblas3 libbullet-dev libbullet2.87 libbz2-dev libccd-dev libccd2 libcharls1 libdap25 libdapclient6v5 libdart6
  libdart6-collision-bullet libdart6-collision-bullet-dev libdart6-collision-ode libdart6-collision-ode-dev libdart6-dev
  libdart6-external-ikfast-dev libdart6-external-odelcpsolver libdart6-external-odelcpsolver-dev libdart6-utils
  libdart6-utils-dev libdart6-utils-urdf libdart6-utils-urdf-dev libdc1394-22 libdc1394-22-dev libegl1-mesa-dev libeigen3-dev
  libepsilon1 libev-dev libev4 libexif-dev libexif-doc libfcl-dev libfcl0.5 libflann-dev libflann1.9 libfreeimage3 libfreexl1
  libfyba0 libgdal20 libgdcm2-dev libgdcm2.8 libgeos-3.6.2 libgeos-c1v5 libgeotiff2 libgflags-dev libgflags2.2 libgfortran4
  libgif7 libgl2ps1.4 libgles2-mesa-dev libglu1-mesa-dev libgpg-error-dev libgpgme-dev libgphoto2-dev libgts-dev libhdf4-0-alt
  libhdf5-100 libhdf5-openmpi-100 libignition-cmake2-dev libignition-common3 libignition-common3-av libignition-common3-av-dev
  libignition-common3-core-dev libignition-common3-dev libignition-common3-events libignition-common3-events-dev
  libignition-common3-graphics libignition-common3-graphics-dev libignition-common3-profiler libignition-common3-profiler-dev
  libignition-fuel-tools3 libignition-gazebo2 libignition-gazebo2-plugins libignition-gui2 libignition-gui2-dev
  libignition-launch libignition-math6 libignition-math6-dev libignition-math6-eigen3-dev libignition-msgs4 libignition-msgs4-dev
  libignition-physics libignition-physics-core-dev libignition-physics-dartsim libignition-physics-dartsim-dev
  libignition-physics-dev libignition-physics-mesh-dev libignition-physics-sdf-dev libignition-plugin libignition-plugin-dev
  libignition-rendering2 libignition-rendering2-core-dev libignition-rendering2-dev libignition-rendering2-ogre1
  libignition-rendering2-ogre1-dev libignition-rendering2-ogre2 libignition-rendering2-ogre2-dev libignition-sensors2
  libignition-sensors2-air-pressure libignition-sensors2-air-pressure-dev libignition-sensors2-altimeter
  libignition-sensors2-altimeter-dev libignition-sensors2-camera libignition-sensors2-camera-dev libignition-sensors2-core-dev
  libignition-sensors2-depth-camera libignition-sensors2-depth-camera-dev libignition-sensors2-dev libignition-sensors2-gpu-lidar
  libignition-sensors2-gpu-lidar-dev libignition-sensors2-imu libignition-sensors2-imu-dev libignition-sensors2-lidar
  libignition-sensors2-lidar-dev libignition-sensors2-logical-camera libignition-sensors2-logical-camera-dev
  libignition-sensors2-magnetometer libignition-sensors2-magnetometer-dev libignition-sensors2-rendering
  libignition-sensors2-rendering-dev libignition-sensors2-rgbd-camera libignition-sensors2-rgbd-camera-dev libignition-tools-dev
  libignition-transport7 libignition-transport7-core-dev libignition-transport7-dev libignition-transport7-log
  libignition-transport7-log-dev libilmbase-dev libjbig-dev libjpeg-dev libjpeg-turbo8-dev libjpeg8-dev libjs-jquery-ui
  libjsoncpp-dev libjsoncpp1 libjxr0 libkmlbase1 libkmldom1 libkmlengine1 liblapack3 liblept5 liblz4-dev liblzma-dev libminizip1
  libmysqlclient20 libnetcdf-c++4 libnetcdf13 liboctomap-dev liboctomap1.8 libodbc1 libode-dev libode6 libogdi3.2 libogg-dev
  libogre-1.9-dev libogre-1.9.0v5 libogre-2.1 libogre-2.1-dev libopencv-calib3d-dev libopencv-calib3d3.2 libopencv-contrib-dev
  libopencv-contrib3.2 libopencv-core-dev libopencv-dev libopencv-features2d-dev libopencv-features2d3.2 libopencv-flann-dev
  libopencv-flann3.2 libopencv-highgui-dev libopencv-highgui3.2 libopencv-imgcodecs-dev libopencv-imgcodecs3.2
  libopencv-imgproc-dev libopencv-ml-dev libopencv-ml3.2 libopencv-objdetect-dev libopencv-objdetect3.2 libopencv-photo-dev
  libopencv-photo3.2 libopencv-shape-dev libopencv-shape3.2 libopencv-stitching-dev libopencv-stitching3.2 libopencv-superres-dev
  libopencv-superres3.2 libopencv-ts-dev libopencv-video-dev libopencv-video3.2 libopencv-videoio-dev libopencv-videoio3.2
  libopencv-videostab-dev libopencv-videostab3.2 libopencv-viz-dev libopencv-viz3.2 libopencv3.2-java libopencv3.2-jni
  libopenexr-dev libpng-dev libpng-tools libpococrypto50 libpocodata50 libpocodatamysql50 libpocodataodbc50 libpocodatasqlite50
  libpocofoundation50 libpocojson50 libpocomongodb50 libpoconet50 libpoconetssl50 libpocoredis50 libpocoutil50 libpocoxml50
  libpocozip50 libpostproc-dev libpq5 libproj12 libprotobuf-dev libprotobuf-lite10 libprotoc-dev libprotoc10 libqhull7
  libqt5designer5 libqt5help5 libqt5opengl5 libqt5opengl5-dev libqt5quickcontrols2-5 libqt5quickparticles5
  libqt5quicktemplates2-5 libqt5quicktest5 libqt5quickwidgets5 libqt5sql5 libqt5sql5-sqlite libqt5test5 libraw1394-dev
  libraw1394-tools libruby2.5 libsdformat8 libsdformat8-dev libsdl2-2.0-0 libsocket++1 libspatialite7 libsqlite3-dev libsuperlu5
  libswresample-dev libswscale-dev libsz2 libtbb-dev libtesseract4 libtheora-dev libtiff-dev libtiff5-dev libtiffxx5
  libtinyxml-dev libtinyxml2-6 libtinyxml2-dev libtinyxml2.6.2v5 liburdfdom-dev liburdfdom-headers-dev liburdfdom-model
  liburdfdom-model-state liburdfdom-sensor liburdfdom-world liburiparser1 libvtk6.3 libwayland-bin libwayland-dev
  libwayland-egl1-mesa libwebsockets8 libwxbase3.0-0v5 libwxgtk3.0-gtk3-0v5 libxerces-c3.2 libyaml-cpp-dev libyaml-cpp0.5v5
  libyaml-dev libzip-dev libzip4 libzmq3-dev libzzip-0-13 mysql-common odbcinst odbcinst1debian2 opencv-data proj-bin proj-data
  protobuf-compiler pyqt5-dev python-asn1crypto python-backports.functools-lru-cache python-cairo python-cffi-backend
  python-cryptography python-cycler python-defusedxml python-enum34 python-gnupg python-idna python-ipaddress python-matplotlib
  python-matplotlib-data python-netifaces python-numpy python-opencv python-opengl python-paramiko python-psutil python-pyasn1
  python-pycryptodome python-pydot python-pyqt5 python-pyqt5.qtopengl python-pyqt5.qtsvg python-pyqt5.qtwebkit
  python-rosdep-modules python-rosdistro python-rosdistro-modules python-rospkg python-rospkg-modules python-sip python-sip-dev
  python-subprocess32 python-tz python-wxgtk3.0 python-wxtools python-wxversion python3-sip python3-sip-dev
  qml-module-qt-labs-folderlistmodel qml-module-qt-labs-settings qml-module-qtgraphicaleffects qml-module-qtqml-models2
  qml-module-qtquick-controls qml-module-qtquick-controls2 qml-module-qtquick-dialogs qml-module-qtquick-layouts
  qml-module-qtquick-privatewidgets qml-module-qtquick-templates2 qml-module-qtquick-window2 qml-module-qtquick2 qt5-qmake
  qt5-qmake-bin qtbase5-dev qtbase5-dev-tools qtchooser qtdeclarative5-dev qtquickcontrols2-5-dev rake ros-melodic-actionlib-msgs
  ros-melodic-angles ros-melodic-bond ros-melodic-bond-core ros-melodic-bondcpp ros-melodic-bondpy ros-melodic-cmake-modules
  ros-melodic-common-msgs ros-melodic-control-msgs ros-melodic-controller-interface ros-melodic-cv-bridge
  ros-melodic-eigen-conversions ros-melodic-geographic-msgs ros-melodic-gl-dependency ros-melodic-hardware-interface
  ros-melodic-joint-state-publisher ros-melodic-kdl-conversions ros-melodic-map-msgs ros-melodic-mav-msgs
  ros-melodic-media-export ros-melodic-message-filters ros-melodic-mk ros-melodic-nav-msgs ros-melodic-orocos-kdl
  ros-melodic-python-orocos-kdl ros-melodic-python-qt-binding ros-melodic-qt-dotgraph ros-melodic-qt-gui
  ros-melodic-qt-gui-py-common ros-melodic-qwt-dependency ros-melodic-ros ros-melodic-ros-environment ros-melodic-ros-ign-bridge
  ros-melodic-ros-tutorials ros-melodic-rosbag-migration-rule ros-melodic-rosbash ros-melodic-rosboost-cfg ros-melodic-rosclean
  ros-melodic-rosconsole-bridge ros-melodic-roscpp-core ros-melodic-roscpp-tutorials ros-melodic-roscreate ros-melodic-rosgraph
[... cut ...]
  ros-melodic-trajectory-msgs ros-melodic-turtlesim ros-melodic-urdf-parser-plugin ros-melodic-urdfdom-py ros-melodic-uuid-msgs
  ros-melodic-visualization-marker-tutorials ros-melodic-visualization-msgs ros-melodic-webkit-dependency ros-melodic-xacro ruby
  ruby-did-you-mean ruby-minitest ruby-net-telnet ruby-power-assert ruby-test-unit ruby2.5 rubygems-integration sbcl
  sdformat8-sdf sip-dev
```
Are there any ways to improve SubT design or installation procedure, in order to avoid this in the future?

nkoenig, 2020-06-09:
This issue has to do with different version of `libcurl`. `npm` wants `libcurl4-gnutls` and `ros` + `ignition` requires `libcurl4-openssl`. In debian you can't have the two version of `libcurl` installed side-by-side, unless you build and install them from source.

The solution to this problem doesn't lie in `subt`. We'd have to change ROS and Ignition dependencies, or convince `npm` to use `libcurl4-openssl`.

pauljurczak, 2020-06-09:
Thanks. I suspected it's not trivial. I can work around it.

closed 2020-06-09 by pauljurczak