osrf/subt #469: Slightly modified SSCI_X4_SENSOR_CONFIG_1 camera position

rsawtell, 2020-06-19:
Would it be possible to either modify this robots configuration, or create a variant of it, in which the camera position is adjusted slightly? Right now the camera is facing directly forward and is placed directly in front of the lidar. This ensures there is a lidar deadzone directly in the middle of the cameras field of view such that approaching an artifact there is no lidar data available whatsoever.

Ideally, it would be out of the way of the lidar (in front of the robot, or maybe just closer to the body so the deadzone isn't as big) and maybe slightly angled down so it has a bit better view of the floor (maybe 10-20 degrees).

pauljurczak, 2020-06-21:
By the same token, it would be nice to be able to change mounting orientation or maybe even mounting location of sensors on any robot on the list, without making a pull request to add the modification. I don't mean to add, remove or replace sensors, just change their mounting.

angelacmaio, 2020-06-25:
Adjusting sensor placement would in effect be a new sensor configuration, and we did not provide a sensor configuration submission window for Cave Circuit (noted in [this post](https://community.subtchallenge.com/t/cost-of-modified-robot-configuration/1255/4)). However, you may want to check out the [new robot models](https://community.subtchallenge.com/t/new-systems-competition-robot-models/1371) to see if those suit your needs.

closed 2020-06-25 by angelacmaio