osrf/subt #476: Description of VGA wide RGBD camera missing

pauljurczak, 2020-06-24:
`ROBOTIKA_X2_SENSOR_CONFIG_1` uses 640 x 360 RGBD camera, which is not on the list at https://github.com/osrf/subt/wiki/api. If I remember correctly, similar camera was listed on the old website.

zwn, 2020-06-24:
It should have been wide camera (Intel Realsense D435) but the horizontal FOV was changed to match the default RGBD camera available in the simulation. So the only effect is that the camera has smaller vertical FOV than the default camera, making it - how to say it - maybe "less useful" is the term I am looking for :worried:. But no sensor updates are allowed for cave so... :-/

pauljurczak, 2020-06-24:
What I meant is that 640x360 RGBD camera **was** defined in old bitbucket wiki. I think it got lost in the move to github.

jisa, 2020-06-24:
We tried to model a real camera that we have in our robots to make it easier to transfer our solutions between Virtual Track and System Track. However, some changes after our initial submit made the camera unlike anything else:

- https://github.com/osrf/subt/commit/2f3f611f5ee71a8c450e607ed54ca92afc0bf9cf
- https://github.com/osrf/subt/commit/575ac447f99be549f67a400fdf0e6897e524d2d4

As far as I remember, now such camera was listed even on the old website. The restrictions to field of view, especially the vertical one, make us struggle.

zwn, 2020-06-24:
> Description of VGA wide RGBD camera missing

What I meant to say is that the camera is not **wide**, its not even regular, its actually quite narrow (vertically).

pauljurczak, 2020-06-24:
@jisa Maybe it was 640x360 RGBD camera with different parameters than yours. The only parameters listed are resolution, FOV and range. Is your FOV 60 deg. and range 10 m?

pauljurczak, 2020-06-24:
@zwn By `wide` I meant wider than 4:3 ratio of VGA. It follows the naming used for sensor specifications, as in `VGA wide camera | 640x360 130-deg 2D camera`

angelacmaio, 2020-06-24:
Now that we have an ever-growing list of sensors, those with slight variations of specs were noted as approximately equivalent to the others in the list. For example, the "640x360 RGBD camera" was removed from the wiki as it is approximated by the "VGA RGBD camera" (640x480). 

This list is meant to give a high-level overview of the sensors available. See this caveat in the wiki:
> **Note: Some team-submitted models have slightly different sensor specifications to better match real sensors. Details can be found in each robot's SubT Tech Repo item and the submitted_models/<ROBOT_NAME>/specifications.md file.

pauljurczak, 2020-06-24:
@angelacmaio Fair enough. The list of robots grew a lot.

closed 2020-06-24 by pauljurczak