osrf/subt #481: Compile error about -fPIC flag

shakeebbb, 2020-07-02:
I'm following the wiki instructions to build the repo. I get the following error during compilation. It's related to compiler needing Position Independent Code flag. Has anyone else faced or solved this issue?

[ 50%] Built target rosgraph_msgs_generate_messages_cpp
[ 50%] Built target roscpp_generate_messages_lisp
[ 50%] Built target roscpp_generate_messages_cpp
[ 50%] Built target rosgraph_msgs_generate_messages_eus
[ 50%] Built target rosgraph_msgs_generate_messages_lisp
[ 50%] Built target rosgraph_msgs_generate_messages_nodejs
[ 50%] Linking CXX shared library /home/bprl/subt_ws/devel/lib/libsubt_communication_client.so
/usr/bin/ld: /usr/local/lib/libprotobuf.a(arena.o): relocation R_X86_64_TPOFF32 against symbol `_ZN6google8protobuf5Arena13thread_cache_E' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libprotobuf.a(common.o): relocation R_X86_64_PC32 against symbol `stderr@@GLIBC_2.2.5' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status

nkoenig, 2020-07-06:
Can you include a link to the specific instructions that you were following?

Also, what OS and gcc version are you using?

nkoenig, 2020-07-17:
Closing due to lack of response.

closed 2020-07-17 by nkoenig