osrf/subt #486: Empty log from Cloudsim run

dsquez, 2020-07-16:
We uploaded a submission to Cloudsim on the cave qualification map, and it shows as terminated with no errors, but when we download the logs, the .tar file is empty. Is there a way to reclaim the log files? The submission id is bf4c66d2-4e3a-4473-857f-4302b942c744. Thank you.

dsquez, 2020-07-20:
I'm trying to playback some other logs and they seem to be missing information. I get this message when playing back locally:

```
[Msg] End of log file reached. Time: Database has no messages
0 seconds
```
When I seek to a specific timestamp I get errors similar to this:
```
[Err] [Conversions.cc:1183] Attempting to convert an IMU sensor message, but the message does not have an IMU nested message.
[Err] [Conversions.cc:1217] Attempting to convert a lidar sensor message, but the message does not have a lidar nested message.
[Err] [Conversions.cc:1217] Attempting to convert a lidar sensor message, but the message does not have a lidar nested message.
[Err] [Conversions.cc:1217] Attempting to convert a lidar sensor message, but the message does not have a lidar nested message.
[Err] [Conversions.cc:1217] Attempting to convert a lidar sensor message, but the message does not have a lidar nested message.
[Err] [Conversions.cc:1217] Attempting to convert a lidar sensor message, but the message does not have a lidar nested message.
[Err] [Conversions.cc:1022] Attempting to convert a magnetometer sensor message, but the message does not have a magnetometer nested message.
[Err] [Conversions.cc:1104] Attempting to convert an air pressure sensor message, but the message does not have an air pressure nested message.
[Err] [Conversions.cc:1183] Attempting to convert an IMU sensor message, but the message does not have an IMU nested message.
```
And the playback window seems to move to the right timestamp, but the robots aren't moving.

I have multiple submissions behaving this way. One example is submission ID e64453f0-2d10-48d4-b454-0b60809a5568

nkoenig, 2020-07-20:
`bf4c66d2-4e3a-4473-857f-4302b942c744` seems to have the logs from the robots, but the simulation log file is only 29 bytes. There is not way to reclaim the simulation log file. I'll re-run your submission and monitor it's progress.

I was able to playback the log file from `e64453f0-2d10-48d4-b454-0b60809a5568`. I ran `ign gazebo -v 4 --playback <directory containing state.tlog>`. What command are you running?

dsquez, 2020-08-04:
I am running the same command with the `-r` flag as well, but I think my issue with `e64453f0-2d10-48d4-b454-0b60809a5568` was I had switched to ignition-citadel to work with the communication breadcrumbs and had not yet switched back to blueprint.

nkoenig, 2020-08-17:
I'm going to close this issue in order to triage the issue tracker. We are monitoring log file uploads.

closed 2020-08-17 by nkoenig