osrf/subt #488: Communication strength with breadcrumbs

dsquez, 2020-07-16:
We have started experimenting with the breadcrumb functionality, and we are surprised by the apparent model for determining communication strength. New breadcrumbs appear to expand signal along line of sight (which makes sense), but this seems to be limited to within a certain radius of the staging area (or wherever the reference vehicle is when using the instructions provided in the wiki), which isn't what we would expect when setting up a communication system. I have created a google slides presentation that outlines where we put breadcrumbs and the subsequent changes in signal on Simple Cave 2. Does this follow the intended behavior, or might there be a bug?

https://docs.google.com/presentation/d/1tyE6HdsrkdCDjlSVKqXAX9lrhFQW6nzWBoz35OolC9Q/edit?usp=sharing

caguero, 2020-07-31:
Hi @dsquez , thanks for the great report. We're right now trying to reproduce your results. We'll go back to you soon.

caguero, 2020-08-03:
@dsquez , I can confirm that we found an issue with breadcrumbs. We're working on the fix.

malcolmst, 2020-09-09:
Just curious if there is any update on this issue? I haven't been experimenting much with the breadcrumbs yet but I see the PR for the fix is still open

nkoenig, 2020-09-11:
The PR is still open due to qualification code-freeze. We will be integrating communication updates next week.

malcolmst, 2020-09-13:
Makes sense, thanks for the update!

DalInar, 2020-09-25:
Hello! I've been experimenting with the breadcrumbs now that the updated comms model has been merged into master (#579, also related to #587 ), and it's behaving a bit differently than I (perhaps naively) expected.  Based on the new changes to the comms model, could you answer the following question:

What is the intended behavior of the RSSI value returned by the `getNeighbors` function when breadcrumbs are deployed?  In particular, say I have a TEAMBASE stationed in the starting area. Should I expect the RSSI between a robot and TEAMBASE to suddenly improve when a breadcrumb is deployed, because the RSSI is based on the connection between the breadcrumb and the robot, or should I simply expect the RSSI to fall off less quickly as I move away from the breadcrumb because the RSSI is somehow based on the entire path back to the starting area?

My simple experiments today (just manually moving a robot into the tunnel, manually deploying breadcrumbs, and then monitoring the RSSI returned by `getNeighbors`) seems to imply that the answer is the latter, but I was hoping for the former :-)

malcolmst, 2020-09-30:
Agree that RSSI to the nearest breadcrumb would be nice information to have (would likely be the only RSSI information easily available in the real world)

acschang, 2020-10-14:
Documentation regarding the revisions to the communication model has been updated on the [wiki](https://github.com/osrf/subt/issues/587#issuecomment-708368217).

An example of how breadcrumbs function under similar conditions to those originally reported are shown below:
[image]

To clarify for @DalInar: the currently reported RSSI values are based upon the entire path to the noted end-point.  For more discussion around RSSI values please refer to #643.

caguero, 2021-01-18:
I believe the original issue reported by @dsquez has been addressed. I'm closing this issue but follow up on #643 for RSSI problems.

closed 2021-01-18 by caguero