osrf/subt #502: Broadcast and "Error parsing artifact"

m3d, 2020-07-27:
Hello,
I am testing communication with `broadcast` ("subt::kBroadcast") and it seems to me that our "proprietary data" are received also by "base station", which is reporting
```
[Err] [BaseStationPlugin.cc:65] Error parsing artifact
```
is it expected?
I am using the latest release 2020-07-23.
Thanks
   Martin/Robotika Team

m3d, 2020-07-27:
p.s. note, that this is happening only near the staging area where the base station is visible to the robot radio ...

m3d, 2020-08-03:
I would consider this as a bug, which is quite bad when we are using our own "TEAMBASE" (which is in contact with "base_station" all the time). Test with reporting sim time as messages from teambase actually generated several false artifacts:
```
0:00:31.612915 23 [b'T60F0R', b'34']
0:00:31.628675 23 [b'base_station', b'\x08\x01\x12']
0:00:32.652941 23 [b'T60F0R', b'35']
...
0:01:19.907739 23 [b'T60F0R', b'80']
0:01:19.980697 23 [b'base_station', b'\x08\x01\x12']
0:01:20.948928 23 [b'T60F0R', b'81']
0:01:20.983311 23 [b'base_station', b'\x08\x01\x12']
0:01:22.004393 23 [b'T60F0R', b'82']
0:01:22.075205 23 [b'base_station', b'\x08\x01\x12']
0:01:23.034794 23 [b'T60F0R', b'83']
0:01:23.092202 23 [b'base_station', b'\x08\x01\x12']
0:01:24.105772 23 [b'T60F0R', b'84']
0:01:24.184011 23 [b'base_station', b'\x08\x01\x12']
0:01:25.139514 23 [b'T60F0R', b'85']
0:01:25.195553 23 [b'base_station', b'\x08\x01\x12']
0:01:26.188951 23 [b'T60F0R', b'86']
0:01:26.282426 23 [b'base_station', b'\x08\x01\x12']
0:01:27.232236 23 [b'T60F0R', b'87']
0:01:27.290824 23 [b'base_station', b'\x08\x01\x12']
0:01:28.272241 23 [b'T60F0R', b'88']
0:01:28.289562 23 [b'base_station', b'\x08\x01\x12']
0:01:29.331475 23 [b'T60F0R', b'89']
0:01:29.383436 23 [b'base_station', b'\x08\x01\x12']
0:01:30.403493 23 [b'T60F0R', b'90']
```
And this is console for simulation:
[image]

caguero, 2020-08-04:
Hi @m3d , if I understand correctly, you're using broadcast as the destination of your messages and they're received by the base station when robots are within range of communication. If that's the case this is the expected behavior of broadcasting messages.

Alternatively, you could use unicast where you specify the name of your robot as destination. Another option is to use `kMulticast` if you `bind` this address to all of your robots previously. Then, the base station won't receive your messages but your entire team of robots will do it (based on range).

m3d, 2020-08-04:
Hi @caguero - yes, correct. Does it mean that you can "broadcast" only artifact messages? Or is there some frame structure which would base station accept but ignore?

caguero, 2020-08-04:
In practice, using the default port the answer is yes. See my previous messages for alternatives (unicast, multicast or broadcast on other ports).

m3d, 2020-08-04:
OK, thanks for hints - I will try different port for broadcast as the first step.
I am closing this issue.

closed 2020-08-04 by m3d