osrf/subt #540: Unification of robotic platforms?

m3d, 2020-08-09:
Hello,
is there plan/time slot for revision already existing platforms? I tried new alternatives but the integration failed due to missing `robot_description` [1] or RGBD camera published only 3D points (not depth image), or RGB camera does not publish compressed (JPEG) image. I could check it again after "Cave Release 6", but maybe it was intentional to publish only what the creator team needed? I am also not sure if publishing extra topics slows down the simulation, i.e. if it is processed "on demand" only?

Thank you
   Martin

[1]

```
EXPLORER_DS1_SENSOR_CONFIG_1

No such file or directory: /home/developer/subt_ws/install/share/explorer_ds1_sensor_config_1/urdf/model.xacro
RLException: Invalid <param> tag: Cannot load command parameter [X4F10R/robot_description]: command [['/opt/ros/melodic/lib/xacro/xacro', '/home/developer/subt_ws/install/share/explorer_ds1_sensor_config_1/urdf/model.xacro', 'name:=X4F10R']] returned with code [2].

Param xml is <param command="$(find xacro)/xacro '$(find explorer_ds1_sensor_config_1)/urdf/model.xacro' name:=$(arg name)" name="$(arg name)/robot_description"/>
The traceback for the exception was written to the log file
```

m3d, 2020-08-10:
p.s. @zwn commented that newly added `optical` section maybe should address this issue?
```
root@gpu-desktop:/osgar-ws# rostopic list
/X4F10R/cmd_vel
/X4F10R/front/camera_info
/X4F10R/front/optical/camera_info
/X4F10R/front/optical/depth
/X4F10R/front/optical/image_raw
/X4F10R/front_scan
/X4F10R/gas_detected
/X4F10R/imu/data
/X4F10R/output/camera_info
/X4F10R/pose
/X4F10R/pose_static
/X4F10R/rgbd_camera/depth/points
/X4F10R/top_scan
/X4F10R/velocity_controller/enable
/clock
/rosout
/rosout_agg
/subt/score
/tf_static
```
(SSCI X4 config 2)
... one comment was also that the topics are not available immediately and may show up "later on". It is not obvious when the robot sensor set is complete & ready.

m3d, 2020-08-10:
p.s.2 I probably mixed up some older notes - if I am patient I get for `SSCI_X4_SENSOR_CONFIG_2`:
```
root@gpu-desktop:/osgar-ws# rostopic list
/X4F10R/air_pressure
/X4F10R/battery_state
/X4F10R/bottom_scan
/X4F10R/cmd_vel
/X4F10R/front/camera_info
/X4F10R/front/depth
/X4F10R/front/depth/compressed
/X4F10R/front/depth/compressed/parameter_descriptions
/X4F10R/front/depth/compressed/parameter_updates
/X4F10R/front/depth/compressedDepth
/X4F10R/front/depth/compressedDepth/parameter_descriptions
/X4F10R/front/depth/compressedDepth/parameter_updates
/X4F10R/front/depth/theora
/X4F10R/front/depth/theora/parameter_descriptions
/X4F10R/front/depth/theora/parameter_updates
/X4F10R/front/image_raw
/X4F10R/front/image_raw/compressed
/X4F10R/front/image_raw/compressed/parameter_descriptions
/X4F10R/front/image_raw/compressed/parameter_updates
/X4F10R/front/image_raw/compressedDepth
/X4F10R/front/image_raw/compressedDepth/parameter_descriptions
/X4F10R/front/image_raw/compressedDepth/parameter_updates
/X4F10R/front/image_raw/theora
/X4F10R/front/image_raw/theora/parameter_descriptions
/X4F10R/front/image_raw/theora/parameter_updates
/X4F10R/front/optical/camera_info
/X4F10R/front/optical/depth
/X4F10R/front/optical/image_raw
/X4F10R/front_scan
/X4F10R/gas_detected
/X4F10R/imu/data
/X4F10R/magnetic_field
/X4F10R/output/camera_info
/X4F10R/pose
/X4F10R/pose_static
/X4F10R/rgbd_camera/depth/points
/X4F10R/top_scan
/X4F10R/velocity_controller/enable
/clock
/rosout
/rosout_agg
/statistics
/subt/score
/tf
/tf_static
```

nkoenig, 2020-08-10:
If you notice missing topics or other information, then please let us know.