osrf/subt #548: Report successful artifact score on retry

dan-riley, 2020-08-14:
Would it be possible to change the ArtifactScore generated by the base_station to always report that it was a scored artifact, instead of only the first time?  This is the scenario I'm seeing:  an artifact is reported and scored successfully, but the response packet is dropped for whatever reason.  So the report is sent again until the ACK is received.  But since it's a repeat, the ACK says the score is zero.  Now, a subsequent robot identifies the same artifact, but with slightly different coordinates.  If we received the original ACK with the score change, we are able to identify internally this is the same artifact and don't try to report it (and therefore wasting one of our report attempts).  However, in the scenario described, the ACK we received said the score was zero, so we have to assume it was not a successful report, and so we report again.  The coordinates have changes, so the scoring counts this as one of our report attempts (I assume...I haven't found a "reports remaining" published like we have in systems?), and again we get an ACK that says zero score.  Even worse, if more robots identify it, they will continue to keep reporting it even though they don't need to. 

We could just not attempt to report those since it is highly possible it's a false positive on our end, but it could also be a positioning error, where the first robot was just outside the 5m bubble, and when the second (or later) robot identifies it, it's now inside.  So I'd prefer to report again when the score was zero, but it's definitely a wasted attempt to report when that artifact was already scored.

Two alternatives:  don't ever drop packets when the ACK is sent, or don't count multiple attempts against a successfully scored artifact as one of our attempts.  Thanks!

nkoenig, 2020-08-24:
We have implemented a change in PR #572. Details on the change can be found in that PR.

dan-riley, 2020-08-24:
Perfect, thanks!

nkoenig, 2020-08-25:
The pull request has been merged into the master branch.

closed 2020-08-25 by nkoenig