osrf/subt #552: Cloudsim environment: X11 server?

peci1, 2020-08-18:
I was quite confused by behavior of the `headless` parameter. It only skips GUI, but rendering still needs OGRE 2. And OGRE 2 in turn needs an X server (or at least it seems so). I'm not actually sure it really is needed for offscreen rendering, but let's leave that for another issue.

The provided docker images do not run or install an X server. So is it supposed that the host machine running the simulation container runs its own X server? How is that achieved on the AWS cloud? Is it automatically provided by the host OS? And what can we do if our own cluster doesn't provide an X server? Would it be possible to add `xvfb` to the sim docker image to get at least something?

Could you add a note about the requirement for an X server to https://github.com/osrf/subt/wiki/Cloudsim%20Docker%20Compose ?

nkoenig, 2020-08-18:
Have you tried using the provided [run.bash](https://github.com/osrf/subt/blob/master/docker/run.bash) script?

peci1, 2020-08-18:
That wouldn't help. I need to run the image using singularity, not docker (but that's a side problem). However, the run.bash script only makes existing X server accessible, and does not start a new one. But we have no X server running on our infrastructure.

nkoenig, 2020-08-18:
I'm not familiar with singularity. I also suggest that you stick to tools that fall within the scope of the SubT Simulator.

You will need to have a valid X server on a machine that is running simulation. We do this on Cloudsim, and you will need to do something similar if you are attempting to re-create Cloudsim locally.

peci1, 2020-08-18:
That is why I'm asking to include xvfb in the docker image. That allows running the cloudsim even on systems without a native X server running. Including this virtual X server doesn't break any existing code and allows people to start the virtual X server if they don't have the real one available.

This could even be followed by a bigger PR that would actually try to detect a running X server, and if none is found, run the virtual one. But that can probably wait after cave...

nkoenig, 2020-08-18:
Sorry for the confusion, but when I said "You will need to have a valid X server on a machine that is running simulation" I mean that this is something that needs to be done outside of the official docker images. This is a system infrastructure issue.

It's been a while since I tried xvfb, but my understanding is that it will not work (or work correctly) with simulation. Have you tried  xvfb, and also collected data from cameras, and RGBD sensor data?

peci1, 2020-08-19:
I agree that it's an infrastructure issue, but we're kind of locked out of all options how to run an X server on our grid. There's just no Xorg and it can't be run without root (which we as users don't have).

I tried xvfb and it works fine rendering-wise - cameras, gpu lidar, everything works. The downside is that rendering is done in software, though...

PR #553 (2020-09-18): Add xvfb to sim image
Partially resolves #552.
closed 2020-09-18 by nkoenig