osrf/subt #554: Interpreting AWS logs

merylspencer, 2020-08-18:
When running tests through the AWS cloud sim environment we can download "get logs" which contains a pos-data folder with files <robotname>-pos.data for each robot in the run. These logs contain data like: 

...
182 368000000 8.45835 -0.839114 0.063492
182 996000000 9.16501 -1.55494 0.063498
183 512000000 10.0475 -2.03782 0.0635
184 28000000 10.9487 -2.47213 0.077411
184 552000000 11.8568 -2.90503 0.124072
185 712000000 12.8506 -3.0521 0.136332
186 228000000 13.6837 -2.49606 0.113003
186 748000000 14.449 -1.84647 0.093469
187 256000000 15.2118 -1.18968 0.093809
...

How is this data meant to be interpreted? I  assume the last three columns are x,y,z positions. Is this correct? What are the first two columns?

nkoenig, 2020-08-18:
The first column is simulation time seconds and the second column is simulation time nanoseconds. The final three columns are x,y,z.

nkoenig, 2020-08-19:
I've added documentation about this file at https://github.com/osrf/subt/wiki/Cloudsim-Log-Playback#robot-position-data

closed 2020-08-19 by nkoenig