osrf/subt #559: Endurance rating of UAVs

pauljurczak, 2020-08-19:
Both MARBLE_QAV500_SENSOR_CONFIG_1 and EXPLORER_DS1_SENSOR_CONFIG_1 have 10 minutes battery lives as measured with simulation clock. Page https://www.subtchallenge.world/openrobotics/fuel/models/EXPLORER_DS1_SENSOR_CONFIG_1 shows one bar at Endurance (Platform battery life) row vs. two bars on https://subtchallenge.world/openrobotics/fuel/models/MARBLE_QAV500_SENSOR_CONFIG_1 page. I'm I missing something?

pauljurczak, 2020-09-21:
I would like to verify that `QAV500` and `DS1` platforms will have 10 minutes battery life as opposed to 20 minutes for `X3` and `X4` for Cave Circuit. It would help to clarify it in https://github.com/osrf/subt/wiki/Battery%20Model%20Tutorial.

angelacmaio, 2020-09-26:
Yes, `QAV500` and `DS1` both have 10 minutes of battery life. The SubT Tech Repo metadata will be updated.

AravindaDP, 2020-12-02:
It would be also beneficial if important characteristics like Max Speed and Endurance is included in robots wiki as well https://github.com/osrf/subt/wiki/Robots

nkoenig, 2021-01-13:
The https://subtchallenge.world/openrobotics/fuel/models/MARBLE_QAV500_SENSOR_CONFIG_1 model's metadata has been updated.

closed 2021-01-13 by nkoenig

pauljurczak, 2021-01-31:
Page https://subtchallenge.world/openrobotics/fuel/models/SOPHISTICATED_ENGINEERING_X4_SENSOR_CONFIG_1 shows 5 bars for SOPHISTICATED_ENGINEERING_X4 endurance, but AFAIK its endurance is 20 minutes - 2 bars.