osrf/subt #560: WorldControl plugin problem

peci1, 2020-08-19:
On one of our laptops, we did a pretty basic install following the catkin ws installation tutorial.

When we launch any of the .ign launch files, the WorldControl plugin is unable to receive the worldName from the Gui plugin, which results in paused simulation which cannot be unpaused:

```
$ ign launch -v 4 cave_circuit.ign worldName:=simple_cave_01 robotName1:=X1 robotConfig1:=X1_SENSOR_CONFIG_1
...
[GUI] [Msg] Transform mode service on [/gui/transform_mode]
[GUI] [Msg] Record video service on [/gui/record_video]
[GUI] [Msg] Move to service on [/gui/move_to]
[GUI] [Msg] Follow service on [/gui/follow]
[GUI] [Msg] View angle service on [/gui/view_angle]
[GUI] [Msg] Added plugin [3D View] to main window
[GUI] [Msg] Loaded plugin [GzScene3D] from path [/usr/lib/x86_64-linux-gnu/ign-gazebo-2/plugins/gui/libGzScene3D.so]
[GUI] [Dbg] [Application.cc:305] Loading plugin [WorldControl]
[GUI] [Wrn] [WorldControl.cc:105] Ignoring service [/world/simple_cave_01/control], world name different from []. Fix or remove your tag.
[GUI] [Msg] Using world control service [/world//control]
[GUI] [Wrn] [WorldControl.cc:172] Ignoring topic [/world/simple_cave_01/stats], world name different from []. Fix or remove your tag.
Topic [/world//stats] is not valid.
[GUI] [Err] [WorldControl.cc:190] Failed to subscribe to [/world//stats]
[GUI] [Msg] Added plugin [World control] to main window
[GUI] [Msg] Loaded plugin [WorldControl] from path [/usr/lib/x86_64-linux-gnu/ign-gui-2/plugins/libWorldControl.so]
[GUI] [Dbg] [Application.cc:305] Loading plugin [WorldStats]
[GUI] [Wrn] [WorldStats.cc:109] Ignoring topic [/world/simple_cave_01/stats], world name different from []. Fix or remove your tag.
Topic [/world//stats] is not valid.
[GUI] [Err] [WorldStats.cc:131] Failed to subscribe to [/world//stats]
[GUI] [Msg] Added plugin [World stats] to main window
[GUI] [Msg] Loaded plugin [WorldStats] from path [/usr/lib/x86_64-linux-gnu/ign-gui-2/plugins/libWorldStats.so]
[GUI] [Dbg] [GazeboGui.cc:144] Running the GazeboGui plugin.
[GUI] [Msg] Loading plugin [ignition-rendering2-ogre2]
[GUI] [Dbg] [RenderUtil.cc:866] Create scene [scene]
[ WARN] [1597837138.934479279]: /use_sim_time set to true and no clock published.  Still waiting for valid time...
[Dbg] [SimulationFeatures.cc:47] Simulation timestep set to: 0.004
[Msg] Serving scene information on [/world/simple_cave_01/scene/info]
[Msg] Serving graph information on [/world/simple_cave_01/scene/graph]
[Msg] Serving full state on [/world/simple_cave_01/state]
[Msg] Publishing scene information on [/world/simple_cave_01/scene/info]
[Msg] Publishing entity deletions on [/world/simple_cave_01/scene/deletion]
[Msg] Publishing state changes on [/world/simple_cave_01/state]
[Msg] Publishing pose messages on [/world/simple_cave_01/pose/info]
...
```

This is weird because on other machines where we installed the system it works.

It seems that [`gui::worldNames()`](https://github.com/ignitionrobotics/ign-gui/blob/7db2281cf05959a4f4b68acdd3baf24591ca75e5/src/plugins/world_control/WorldControl.cc#L88) does not return any worlds, or returns one with an empty name.

This laptop is different just by not having a NVidia GPU. We use Intel drivers from https://launchpad.net/~oibaf/+archive/ubuntu/graphics-drivers which provide OpenGL 4.6 on the integrated GPU. But I can't see how that could be related to the observed problem. Rendering in the Gui window works normally.

I've tried manually substituting `<%= $worldName %>` in `install/share/subt_ign/launch/cave_circuit.ign` with `simple_cave_01`, but that did not help.

`ign service -s /world/simple_cave_01/control -i` told me there are no providers of this service. `ign topic -t /world/simple_cave_01/stats -i` told me the topic exists.

Would you please help me debugging this issue?

nkoenig, 2020-08-19:
1. Which install instructions did you follow?

2. What version of ign-gazebo do you have installed?

3. Do you have multiple instances of gazebo running?

nkoenig, 2020-08-19:
Found this problem here: https://github.com/ignitionrobotics/ign-gui/pull/107, and reverted here: https://github.com/ignitionrobotics/ign-gui/pull/110.

New debians will be available in about 20 minutes.

nkoenig, 2020-08-19:
Debians are out. You can update using:

```
sudo apt-get update
sudo apt-get upgrade libignition-gui2-dev
```

nkoenig, 2020-08-20:
I've had external confirmation that this issue has been fixed. I'll close this issue.