osrf/subt #577: model.xacro missing for EXPLORER_DS1_SENSOR_CONFIG_1 in subt-virtual-testbed image

pauljurczak, 2020-08-26:
When running the newest `subt-virtual-testbed` image, I'm getting this error:
```
No such file or directory: /home/developer/subt_ws/install/share/explorer_ds1_sensor_config_1/urdf/model.xacro
RLException: Invalid <param> tag: Cannot load command parameter [X1/robot_description]: command [['/opt/ros/melodic/lib/xacro/xacro', '/home/developer/subt_ws/install/share/explorer_ds1_sensor_config_1/urdf/model.xacro', 'name:=X1']] returned with code [2]. 
```
and indeed listing this directory shows no such file:
```
developer@desktop:~/subt_ws/install/share/explorer_ds1_sensor_config_1/urdf$ dir
model.orig.xacro  robot_from_sdf.xacro
```

nkoenig, 2020-08-27:
This was fixed in #570 , and new [docker images](https://github.com/osrf/subt/wiki/release_notes#2020-08-27) have been created.

pauljurczak, 2020-08-28:
Works fine now. Thank you.

closed 2020-08-28 by pauljurczak