osrf/subt #581: /subt/run_clock returns 0 seconds during run

rsawtell, 2020-08-26:
The run_clock topic seems to work fine in the setup phase, but only ever reports 0 seconds once the run phase has begun. Is this intentional or a bug?

>\---
phase: "setup"
data: 
  secs: 853
  nsecs:         0
\---
phase: "setup"
data: 
  secs: 852
  nsecs:         0
\---
phase: "run"
data: 
  secs: 0
  nsecs:         0
\---
phase: "run"
data: 
  secs: 0
  nsecs:         0
\---`

pauljurczak, 2020-08-27:
I noticed the same behavior:
```
solution1_1  | battery: 0.999%  clock: 8.944  timer: 0.  score: 0
solution1_1  | battery: 0.999%  clock: 9.144  timer: 0.  score: 0
solution1_1  | battery: 0.999%  clock: 9.348  timer: 0.  score: 0
solution1_1  | battery: 0.999%  clock: 9.572  timer: 0.  score: 0
solution1_1  | battery: 0.998%  clock: 9.780  timer: 0.  score: 0
solution1_1  | battery: 0.998%  clock: 9.976  timer: 0.  score: 0
solution1_1  | battery: 0.998%  clock: 10.152  timer: 0.  score: 0
solution1_1  | battery: 0.998%  clock: 10.304  timer: 0.  score: 0
solution1_1  | battery: 0.998%  clock: 10.484  timer: 0.  score: 0
solution1_1  | battery: 0.997%  clock: 10.708  timer: 0.  score: 0
solution1_1  | battery: 0.997%  clock: 10.940  timer: 0.  score: 0
solution1_1  | battery: 0.997%  clock: 11.208  timer: 0.  score: 0
solution1_1  | battery: 0.996%  clock: 11.672  timer: 0.  score: 0
solution1_1  | battery: 0.995%  clock: 12.272  timer: 0.  score: 0
solution1_1  | battery: 0.994%  clock: 12.900  timer: 0.  score: 0
solution1_1  | battery: 0.993%  clock: 13.472  timer: 0.  score: 0
```
where `clock` shows `/clock` and `timer` shows `/subt/run_clock`

nkoenig, 2020-08-27:
This is expected behavior. There is no time limit during a local run. You can set `durationSec:=<sim_seconds_allowed>` to give yourself a timelimit.

More information is on the wiki: https://github.com/osrf/subt/wiki/api#competition-clock.

closed 2020-08-27 by nkoenig