osrf/subt #583: Unexpected camera rates for EXPLORER_DS1_SENSOR_CONFIG_1

pauljurczak, 2020-08-27:
`EXPLORER_DS1_SENSOR_CONFIG_1/model.sdf` reads:
```
      <sensor name="rs_up" type="rgbd_camera">
        <pose>0.305 0 0.11 0 -1.57079632679 0</pose>
        <always_on>1</always_on>
        <update_rate>20</update_rate>
```
and topic monitoring also shows update rate around 19Hz, but actual data from the cameras indicate update rate of about 53Hz. Here is the listing from Docker Composer run with each line containing data from headers of 6 images (depth and rgb from each of 3 cameras):
```
{<ROS clock>}  frame_id  [stamp0  ...   stamp5]
```
```
solution1_1  | {8.904}   23  [8.876 8.876 8.876 8.876 8.876 8.876 ]
solution1_1  | {8.908}   24  [8.904 8.904 8.904 8.904 8.904 8.904 ]
solution1_1  | {8.936}   25  [8.908 8.908 8.908 8.908 8.908 8.908 ]
solution1_1  | {8.956}   26  [8.936 8.936 8.936 8.936 8.936 8.936 ]
solution1_1  | {8.960}   27  [8.956 8.956 8.956 8.956 8.956 8.956 ]
solution1_1  | {8.996}   28  [8.960 8.960 8.960 8.960 8.960 8.960 ]
solution1_1  | {9.008}   29  [8.996 8.996 8.996 8.996 8.996 8.996 ]
solution1_1  | {9.012}   30  [9.008 9.008 9.008 9.008 9.008 9.008 ]
solution1_1  | {9.052}   31  [9.012 9.012 9.012 9.012 9.012 9.012 ]
solution1_1  | {9.060}   32  [9.056 9.056 9.056 9.056 9.056 9.056 ]
solution1_1  | {9.064}   33  [9.060 9.060 9.060 9.060 9.060 9.060 ]
solution1_1  | {9.112}   34  [9.064 9.064 9.064 9.064 9.064 9.064 ]
solution1_1  | {9.116}   35  [9.112 9.112 9.112 9.112 9.112 9.112 ]
solution1_1  | {9.120}   36  [9.116 9.116 9.116 9.116 9.116 9.116 ]
solution1_1  | {9.164}   37  [9.120 9.120 9.120 9.120 9.120 9.120 ]
solution1_1  | {9.180}   38  [9.164 9.164 9.164 9.164 9.164 9.164 ]
solution1_1  | {9.212}   39  [9.180 9.180 9.180 9.180 9.180 9.180 ]
solution1_1  | {9.240}   40  [9.212 9.212 9.212 9.212 9.212 9.212 ]
solution1_1  | {9.260}   41  [9.240 9.240 9.240 9.240 9.240 9.240 ]
solution1_1  | {9.264}   42  [9.260 9.260 9.260 9.260 9.260 9.260 ]
solution1_1  | {9.300}   43  [9.264 9.264 9.264 9.264 9.264 9.264 ]
solution1_1  | {9.304}   44  [9.300 9.300 9.300 9.300 9.300 9.300 ]
solution1_1  | {9.312}   45  [9.304 9.304 9.304 9.304 9.304 9.304 ]
solution1_1  | {9.360}   46  [9.312 9.312 9.312 9.312 9.312 9.312 ]
solution1_1  | {9.364}   47  [9.360 9.360 9.360 9.360 9.360 9.360 ]
solution1_1  | {9.368}   48  [9.364 9.364 9.364 9.364 9.364 9.364 ]
solution1_1  | {9.404}   49  [9.368 9.368 9.368 9.368 9.368 9.368 ]
solution1_1  | {9.428}   50  [9.408 9.408 9.408 9.408 9.408 9.408 ]
solution1_1  | {9.432}   51  [9.428 9.428 9.428 9.428 9.428 9.428 ]
solution1_1  | {9.456}   52  [9.432 9.432 9.432 9.432 9.432 9.432 ]
solution1_1  | {9.492}   53  [9.456 9.456 9.456 9.456 9.456 9.456 ]
solution1_1  | {9.496}   54  [9.492 9.492 9.492 9.492 9.492 9.492 ]
solution1_1  | {9.504}   55  [9.496 9.496 9.496 9.496 9.496 9.496 ]
solution1_1  | {9.508}   56  [9.504 9.504 9.504 9.504 9.504 9.504 ]
solution1_1  | {9.548}   57  [9.508 9.508 9.508 9.508 9.508 9.508 ]
solution1_1  | {9.604}   58  [9.556 9.556 9.556 9.556 9.556 9.556 ]
solution1_1  | {9.608}   59  [9.604 9.604 9.604 9.604 9.604 9.604 ]
solution1_1  | {9.616}   60  [9.608 9.608 9.608 9.608 9.608 9.608 ]
solution1_1  | {9.656}   61  [9.616 9.616 9.616 9.616 9.616 9.616 ]
solution1_1  | {9.660}   62  [9.656 9.656 9.656 9.656 9.656 9.656 ]
solution1_1  | {9.676}   63  [9.660 9.660 9.660 9.660 9.660 9.660 ]
solution1_1  | {9.708}   64  [9.676 9.676 9.676 9.676 9.676 9.676 ]
solution1_1  | {9.712}   65  [9.708 9.708 9.708 9.708 9.708 9.708 ]
solution1_1  | {9.736}   66  [9.712 9.712 9.712 9.712 9.712 9.712 ]
solution1_1  | {9.740}   67  [9.736 9.736 9.736 9.736 9.736 9.736 ]
solution1_1  | {9.760}   68  [9.740 9.740 9.740 9.740 9.740 9.740 ]
solution1_1  | {9.800}   69  [9.760 9.760 9.760 9.760 9.760 9.760 ]
solution1_1  | {9.804}   70  [9.800 9.800 9.800 9.800 9.800 9.800 ]
solution1_1  | {9.808}   71  [9.804 9.804 9.804 9.804 9.804 9.804 ]
solution1_1  | {9.852}   72  [9.808 9.808 9.808 9.808 9.808 9.808 ]
solution1_1  | {9.860}   73  [9.856 9.856 9.856 9.856 9.856 9.856 ]
solution1_1  | {9.864}   74  [9.860 9.860 9.860 9.860 9.860 9.860 ]
solution1_1  | {9.868}   75  [9.864 9.864 9.864 9.864 9.864 9.864 ]
solution1_1  | {9.908}   76  [9.868 9.868 9.868 9.868 9.868 9.868 ]
solution1_1  | {9.928}   77  [9.908 9.908 9.908 9.908 9.908 9.908 ]
solution1_1  | {9.956}   78  [9.928 9.928 9.928 9.928 9.928 9.928 ]
solution1_1  | {9.960}   79  [9.956 9.956 9.956 9.956 9.956 9.956 ]
solution1_1  | {9.988}   80  [9.960 9.960 9.960 9.960 9.960 9.960 ]
solution1_1  | {9.992}   81  [9.988 9.988 9.988 9.988 9.988 9.988 ]
```

nkoenig, 2020-08-27:
Can you provide interpretation of your displayed data, and how I can reproduce your setup?

pauljurczak, 2020-08-28:
To make it simpler, here is a test program:
```
#include <iostream>
#include <ros/ros.h>
#include <ros/callback_queue.h>
#include <sensor_msgs/Image.h>

using namespace std;

ros::NodeHandle* nodeHandle;  
ros::Subscriber* imageSubsc;
int   imageId{-1};
float imageStamp{-1};

void imageCallback(const sensor_msgs::Image::ConstPtr& message) {
  imageId    = message->header.seq;
  imageStamp = message->header.stamp.sec + 1e-9 * message->header.stamp.nsec;
}

int main(int argc, char* argv[]) {
  ros::init(argc, argv, "igntest");
  cout << "Node created" << endl;
  ros::Time::init();
  ros::Duration(10.0).sleep();

  nodeHandle = new ros::NodeHandle; // this will create the node and make it visible to ROS 
  imageSubsc = new ros::Subscriber(nodeHandle->subscribe<sensor_msgs::Image>("/X1/front_rgbd/image_raw", 1, imageCallback, ros::TransportHints().tcpNoDelay()));

  ros::Duration(10.0).sleep();

  for (int i = 0; i < 50; i++) {
    ros::getGlobalCallbackQueue()->callAvailable(ros::WallDuration(0.1));
    cout << fixed << setprecision(3) << ros::Time::now().toSec() << "  " << imageStamp << "  " << imageId << endl;
  }
}
```
When executed as a binary with `ign launch -v 1 cave_circuit.ign worldName:=cave_qual robotName1:=X1 robotConfig1:=EXPLORER_DS1_SENSOR_CONFIG_1` running in another process, it produces this output (each line contains ROS time when line displayed, image time stamp and image sequence ID):
```
21.024  21.000  218
21.068  21.052  219
21.136  21.052  219
21.136  21.100  220
21.160  21.152  221
21.240  21.204  222
21.260  21.252  223
21.336  21.252  223
21.340  21.304  224
21.400  21.304  224
21.404  21.352  225
21.404  21.400  226
21.468  21.400  226
21.472  21.452  227
21.536  21.452  227
21.536  21.500  228
21.560  21.552  229
21.652  21.552  229
21.652  21.604  230
21.660  21.652  231
21.736  21.652  231
21.740  21.704  232
21.804  21.752  233
21.808  21.800  234
21.856  21.800  234
21.860  21.852  235
21.936  21.852  235
21.940  21.904  236
22.000  21.952  237
22.008  22.000  238
...
```
which is consistent with 20 FPS (frames per second). The same binary inside docker image run with Docker Composer and the newest SubT images produces:
```
solution1_1  | 13.976  13.956  220
solution1_1  | 14.000  13.960  221
solution1_1  | 14.008  14.004  222
solution1_1  | 14.016  14.008  223
solution1_1  | 14.060  14.012  224
solution1_1  | 14.072  14.060  225
solution1_1  | 14.072  14.068  226
solution1_1  | 14.108  14.072  227
solution1_1  | 14.116  14.108  228
solution1_1  | 14.140  14.112  229
solution1_1  | 14.160  14.136  230
solution1_1  | 14.164  14.160  231
solution1_1  | 14.204  14.164  232
solution1_1  | 14.208  14.204  233
solution1_1  | 14.216  14.208  234
solution1_1  | 14.220  14.212  235
solution1_1  | 14.264  14.216  236
solution1_1  | 14.272  14.264  237
solution1_1  | 14.272  14.268  238
solution1_1  | 14.316  14.272  239
solution1_1  | 14.340  14.312  240
solution1_1  | 14.340  14.336  241
solution1_1  | 14.364  14.340  242
solution1_1  | 14.364  14.360  243
solution1_1  | 14.404  14.364  244
solution1_1  | 14.412  14.404  245
solution1_1  | 14.416  14.412  246
solution1_1  | 14.464  14.416  247
solution1_1  | 14.468  14.464  248
solution1_1  | 14.504  14.468  249
solution1_1  | 14.516  14.512  250
solution1_1  | 14.540  14.516  251
solution1_1  | 14.564  14.536  252
solution1_1  | 14.576  14.564  253
solution1_1  | 14.608  14.568  254
solution1_1  | 14.608  14.604  255
solution1_1  | 14.620  14.608  256
solution1_1  | 14.624  14.616  257
solution1_1  | 14.668  14.620  258
solution1_1  | 14.672  14.668  259
solution1_1  | 14.716  14.672  260
solution1_1  | 14.736  14.720  261
solution1_1  | 14.768  14.736  262
solution1_1  | 14.776  14.768  263
solution1_1  | 14.808  14.772  264
solution1_1  | 14.808  14.804  265
solution1_1  | 14.824  14.808  266
solution1_1  | 14.824  14.820  267
solution1_1  | 14.872  14.824  268
solution1_1  | 14.872  14.868  269
```
which is more than 50 FPS.

pauljurczak, 2020-08-28:
Additional problem with the test above is that catkin build version produces repetitions like:
```
21.068  21.052  219
21.136  21.052  219
```
which are spaced by less than 0.1s, the timeout in:
```
ros::getGlobalCallbackQueue()->callAvailable(ros::WallDuration(0.1));
```
This should not be possible, unless there is a problem with ROS clock.

pauljurczak, 2020-08-30:
I repeated these tests with `X3_SENSOR_CONFIG_5` and frame rates are fine with Docker Compose, but callback timeout is still wrong in catkin build. 

I can deal with callback timeout issue, but the extra frames produced in Cloudsim with `EXPLORER_DS1_SENSOR_CONFIG_1` are problematic for SLAM. Their timestamp spacing is non-linear.

nkoenig, 2021-03-02:
Do you still see this issue? I tested in both a catkin workspace and docker image. Both scenarios produced expected results.

## Catkin workspace

1. `ign launch -v 1 competition.ign worldName:=cave_qual circuit:=cave robotName1:=X1 robotConfig1:=EXPLORER_DS1_SENSOR_CONFIG_1`
2. `rostopic hz /X1/front_rgbd/depth/image_raw`

which produces the following output:

```
average rate: 19.958
	min: 0.036s max: 0.096s std dev: 0.00917s window: 265
average rate: 19.991
	min: 0.036s max: 0.096s std dev: 0.00900s window: 276
```

## Using docker

1. `./run.bash cloudsim_sim cloudsim_sim.ign worldName:=cave_qual circuit:=cave robotName1:=X1 robotConfig1:=EXPLORER_DS1_SENSOR_CONFIG_1`
2. `./run.bash cloudsim_bridge worldName:=cave_qual circuit:=cave robotName1:=X1 robotConfig1:=EXPLORER_DS1_SENSOR_CONFIG_1`
3. `rostopic hz /X1/front_rgbd/depth/image_raw`

which produces the following output
```
average rate: 20.006
	min: 0.000s max: 0.112s std dev: 0.00978s window: 140
average rate: 20.064
	min: 0.000s max: 0.112s std dev: 0.00943s window: 151
```