osrf/subt #587: Trying to understand the RSSI in relation to breadcrumbs

arjo129, 2020-08-30:
So we have been trying to use the RSSI signal of the breadcrumbs to gauge when to drop them. It seems like when a breadcrumb is deployed the RSSI doesn't change. This is contrary to real world experience where we usually measure RSSI based on the nearest meshnode in the network.

peci1, 2020-09-01:
Probably duplicate of #488?

acschang, 2020-09-08:
Related to #488

Please post a specific test scenario for further investigation.

arjo129, 2020-09-10:
Steps to reproduce:

- Move robot forward to a certain position.
- Use `getNeighbours()` to see who is visible and rssi
- Drop beacon
- Use `getNeighbours()` to see who is visible and rssi

RSSI still remains within same range. This is really weird cause in real life RSSI is usually measured by the analog frontend of the radio. This means if I have mesh network it normally would correspond to the signal strength of the nearest mesh relay. Reading through #488 more closely it looks like it may be related.

acschang, 2020-10-14:
The documentation for the communication model has been updated in the [wiki](https://github.com/osrf/subt/wiki/Communication%20Model%20Tutorial.)  In the case of the simulated communication model, RSSI is point-to-point as opposed to the RSSI of the nearest breadcrumb.  Another issue related to RSSI has been reported in #643.