osrf/subt #592: Incomplete bagging of data

rsawtell, 2020-09-04:
I've encountered a problem where sometimes the bag file isn't completed. Instead we only get the

robot_data_1.bag.active

and not

robot_data_1.bag

running rosbag reindex at least allows us to access our bagged data but this seems less than ideal. It is inconsistent even within the same simulation, with one robot returning perfectly good bag files while others have only the .active files.

acschang, 2020-09-08:
Have you been able to determine if any logged data is missing from the reconverted `*.bag.active` files as opposed to the completed `*.bag` files?  An example of this occurring could be timestamped sensor data in the `*.bag` file more than a few seconds beyond that of `*.bag.active` file or segments of lost data throughout the run.

rsawtell, 2020-09-08:
I can't totally confirm or deny if that's happening, as we only log data periodically (when artifacts are seen), but the detections I expect to see there based on reporting are all there. That said, it seems like it's related to the bag file size limit. Eg, if we record less that the file limit we end with the .active file that isn't indexed, and if we go over the first file limit that bag is closed, the second is started, and the second one is the one that remains in .active with no index.

zwn, 2020-09-09:
@rsawtell That is our observation as well. We don't observe any missing data (and we are heavy users of the logs).

mjcarroll, 2020-09-09:
This is likely due to the way that `rosbag record` interacts with the shutdown procedure. The data is copied once the simulation is over, but potentially before `rosbag` is completely shutdown.

In general, as `rosbag` is recording, it is writing incoming messages to disk in chunks (default is 768kb).  While the bag is being actively recorded, it will be suffixed by `.active`.  

When `rosbag record` is stopped gracefully, it will create a set of journal/index data and rename the file to remove the suffix.  
If `rosbag record` isn't stopped gracefully, all data should still be preserved, just without the journal/index.  
The `rosbag reindex` command can reconstruct the index.

In the case that you see multiple bag files, it is because the `record` command has finalized/indexed/closed the first bag and started recording the second one.

TL;DR: The data should be there, you may just have to reindex.

zwn, 2020-09-10:
Can this be related to the fact that bridge container does not respond correctly to Ctrl-C? When I run the bridge container from a terminal and send Ctrl-C to it it starts to shutdown but timeouts and says `escalating to SIGTERM`. The whole process takes about 15s. It would be nicer if it could shutdown itself cleanly.

nkoenig, 2020-11-30:
This has been fixed in PR #708

nkoenig, 2020-12-16:
This has been fixed and released.

closed 2020-12-17 by nkoenig