osrf/subt #594: Rosdep keys for ignition-gazebo2?

peci1, 2020-09-05:
https://github.com/osrf/subt/blob/60aeb723be6af834c1499221f2a7ad16f926321c/submitted_models/ctu_cras_norlab_absolem_sensor_config_1/package.xml#L18-L21

I've just noticed rosdep doesn't know these keys. Should they be added? I can do the rosdep PR, but first I wanted to know if there aren't any other plans regarding Blueprint, like not "messing" rosdep with it when it's almost EOL... But adding the keys would be the cleanest solution.

mjcarroll, 2020-09-09:
Since these packages are not available in the upstream Ubuntu repositories, but only available via `packages.osrfoundation.org`, it doesn't make sense to add them to rosdep.  In this case it would end up with packages that are not able to be resolved without the PPA.  This is explicit design decision on the part of rosdep: https://answers.ros.org/question/70210/does-rosdep-support-installing-dependencies-from-ppas/

As a practice, I favor putting these keys in, because if you are building as part of a larger workspace, it will guarantee that a tool like `colcon` will build the packages in the correct order to resolve dependencies.  Any rosdep warnings around these keys can be safely ignored.

peci1, 2020-09-09:
Okay, so you suggest to leave everything as is?

azeey, 2020-09-30:
Yes, I think that's what @mjcarroll is suggesting.

closed 2020-09-30 by azeey