osrf/subt #597: Help required with modelling Realsense T265 tracking camera

mihirk284, 2020-09-09:
I am trying to submit a model of our flying robot for Team CERBERUS. We have a Realsense T265 tracking camera onboard the robot. Our robot will have only the following sensors T265, IMU, 2x ( single ray ToF sensors). This tracking camera is also the sensor that is the source of onboard odometry for the robot.

I am not able to find the best way to model the fisheye lens of the camera. Can someone please guide me on how I can model the sensor? Also since this sensor will be an onboard source of Odometry of the robot, will it be okay to enable ground truth as a default for this robot?

angelacmaio, 2020-09-11:
Please submit the model with only the raw sensor outputs (not fused odometry) published so developers can add their own SLAM solutions to the robot. Ground truth can always be enabled during local testing by passing `enableGroundTruth:=true` to the `ign launch` argument, but ground truth will not be available when the robot is used in competition.

@nkoenig do you have any suggestions for fisheye lens modeling?

nkoenig, 2020-09-11:
What is the field of view of the fish eye lens?

mihirk284, 2020-09-11:
Hi @nkoenig 
The FoV as mentioned on their website is:-
>The camera includes two OV9282 imagers with fisheye lenses for a combined, close to hemispherical 163±5° field of view for robust tracking even with fast motion.

The product link can be found here is [here](https://www.intelrealsense.com/tracking-camera-t265/).

nkoenig, 2020-09-16:
@iche033 , have we ported wide angle camera sensors to Ignition?

iche033, 2020-09-16:
no, fisheye / wide angle cameras are not supported in ignition yet.

bfotheri, 2021-02-16:
Bump, are there are any plans for porting the wide angle lens plugin? Or doing something similar to support wide angle lenses?