osrf/subt #600: How can the solution code learn the robot configuration?

zwn, 2020-09-10:
Is there some `rosparam` or `rostopic` that would list this information? If not, can it be added?

nkoenig, 2020-09-11:
You can communicate this type of information through the comms model.

Alternatively, if you know the robots that you want to use ahead of time, then you can bake this information into the robot docker images.

zwn, 2020-09-14:
> You can communicate this type of information through the comms model.

That is a bit awkward for us since the startup of our solution needs this information and comms is not available for us in that part of our code. As we learn in #270 we cannot do a single shot communication along the lines of calling `rosparam` or `rostopic` so that further complicates things. We would have to get comms up and running before the rest of the system which is something we are not ready to do at this time.

> Alternatively, if you know the robots that you want to use ahead of time, then you can bake this information into the robot docker images.

That is the point of my question - we would like to have a single docker image supporting multiple configuration since that works better for us - logistics-wise.

Would it be possible to add some `rosparam`-able or `rostopic`-able way to learn the robot configuration? Some kind of introspection.

zwn, 2020-09-15:
> You can communicate this type of information through the comms model.

I just realized I cannot. The information is just not available so there is nothing to send through the comms model. Hard coding it is the only possible way right now.

AravindaDP, 2020-10-30:
I see that we could parse `/<robot_name>/robot_description` rosparam to get some information regarding robot configuration but I guess this could be a fragile approach. (Besides for some reason this ros param is not available when I use catkin_install launch command)

@nkoenig Now that the information we need is available on simulation container according to https://github.com/osrf/subt/pull/584 can we have same information available as rosparam on solution container?