osrf/subt #604: Multicast over SubT comms is not allowed

tpet, 2020-09-11:
I am not able to bind to a multicast group.
E.g. the code below (excerpt from within a member function)
```
comms_client_ = std::make_shared<subt::CommsClient>("X1");
// subt::communication_broker::kMulticast == "multicast"
comms_client_->Bind(&SimpleRelay::comms_callback, this, subt::communication_broker::kMulticast, 20) // returns false
```
fails with
`[ERROR 1599819990.746046087 /subt_ros_relay] OnBind address does not match origination. Attempted impersonation of a robot as robot[multicast].`

I'm using an image based on osrf/subt-virtual-testbed:latest.

Is multicast supported for cave circuit? From [the comms client docs](https://github.com/osrf/subt/blob/master/subt-communication/subt_communication_broker/include/subt_communication_broker/subt_communication_client.h#L100) it seems that it should be. (`"kMulticast"` doesn't work either.)

AravindaDP, 2020-10-17:
@nkoenig @caguero Any updates on this? I also came across same issue and couldn't figure out how to properly use multicast functionality. Could there be a bug in underline CommsClient?

peci1, 2021-02-01:
Solution proposed in #772.