osrf/subt #607: Ground truth pose publishers should not publish static TFs

peci1, 2020-09-14:
In all robot models, e.g.

https://github.com/osrf/subt/blob/9b8c41f85282c8b68f278dfc0a47e35148ba994b/submitted_models/ctu_cras_norlab_absolem_sensor_config_1/launch/spawner.rb#L107-L109

the ground truth pose is published as a static TF. It works, but it's not right, because static TFs should only be used for transforms that do not change or change only occasionally.

I think it could be fixed by adding this snippet to all robots:

```xml
    <plugin filename="libignition-gazebo-pose-publisher-system.so"
        name="ignition::gazebo::systems::PosePublisher">
        <publish_link_pose>false</publish_link_pose>
        <publish_sensor_pose>false</publish_sensor_pose>
        <publish_collision_pose>false</publish_collision_pose>
        <publish_visual_pose>false</publish_visual_pose>
        <publish_nested_model_pose>#{$enableGroundTruth}</publish_nested_model_pose>
        <use_pose_vector_msg>true</use_pose_vector_msg>
        <static_publisher>false</static_publisher>
      </plugin>
```

and setting 

```xml
    <publish_nested_model_pose>false</publish_nested_model_pose>
```

in the static transform snippets.

acschang, 2020-10-14:
Related to #659

peci1, 2021-01-19:
Solution proposed in #759.

iche033, 2021-03-05:
the changes look good to me and would resolve the issue that model poses are incorrectly published to the pose_static topic.

This makes me think that the PosePublisher system should really just be classifying a model / nested model entity as a dynamic entity instead of a static one (in which case, we would just insert the entity into the dynamic entities list somewhere around [here](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo4/src/systems/pose_publisher/PosePublisher.cc#L402), i.e. `if (nestedModel) this->dynamicEntities.insert(entity)`)