osrf/subt #612: Segmentation fault for  ROBOTIKA_FREYJA_SENSOR_CONFIG_2

m3d, 2020-09-16:
Hello,
I wanted to try breadcrumbs on robot Freyja (config 2) but it crashes on simulator start (release 2020-09-01, local run)
```
Stack trace (most recent call last) in thread 106:
#29   Object "", at 0xffffffffffffffff, in
#28   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fb6ca225a3e, in clone
#27   Object "/lib/x86_64-linux-gnu/libpthread.so.0", at 0x7fb6c9eec6da, in start_thread
#26   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fb6c53ff17c, in QThread::setTerminationEnabled(bool)
#25   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fb6c53fa239, in QThread::exec()
#24   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fb6c55db909, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#23   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fb6c563688e, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#22   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fb6c18456db, in g_main_context_iteration
#21   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fb6c184564f, in g_main_context_dispatch
#20   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fb6c1845416, in g_main_context_dispatch
#19   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fb6c5637262, in QEventDispatcherGlibPrivate::runTimersOnceWithNormalPriority()
#18   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fb6c55e004c, in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
#17   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fb6c55dd8d7, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#16   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fb6c55dd769, in QCoreApplication::notify(QObject*, QEvent*)
#15   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fb6c560d0c1, in QObject::event(QEvent*)
#14   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-2/plugins/gui/libGzScene3D.so", at 0x7fb692bd17f7, in ignition::gazebo::v2::RenderThread::RenderNext()
#13   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-2/plugins/gui/libGzScene3D.so", at 0x7fb692bccd6d, in ignition::gazebo::v2::IgnRenderer::Initialize()
#12   Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo2-rendering.so.2", at 0x7fb69294e68c, in ignition::gazebo::v2::RenderUtil::Init()
#11   Object "/usr/lib/x86_64-linux-gnu/libignition-rendering2.so.2", at 0x7fb6926eb057, in ignition::rendering::v2::RenderEngineManager::Engine(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#10   Object "/usr/lib/x86_64-linux-gnu/libignition-rendering2.so.2", at 0x7fb6926eae42, in ignition::rendering::v2::RenderEngineManagerPrivate::Engine(std::_Rb_tree_iterator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, ignition::rendering::v2::RenderEngine*> >, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#9    Object "/usr/lib/x86_64-linux-gnu/libignition-rendering2.so.2", at 0x7fb6926f539d, in ignition::rendering::v2::BaseRenderEngine::Load(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)
#8    Object "/usr/lib/x86_64-linux-gnu/libignition-rendering2-ogre2.so", at 0x7fb67a0cb63a, in ignition::rendering::v2::Ogre2RenderEngine::LoadImpl(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocato
#7    Object "/usr/lib/x86_64-linux-gnu/libignition-rendering2-ogre2.so", at 0x7fb67a0cb284, in ignition::rendering::v2::Ogre2RenderEngine::LoadAttempt()
#6    Object "/usr/lib/x86_64-linux-gnu/libignition-rendering2-ogre2.so", at 0x7fb67a0c6cdb, in ignition::rendering::v2::Ogre2RenderEngine::LoadPlugins()
#5    Object "/usr/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7fb679afb312, in Ogre::Root::loadPlugin(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#4    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/RenderSystem_GL3Plus.so", at 0x7fb67816b443, in dllStartPlugin
#3    Object "/usr/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7fb679afb0e5, in Ogre::Root::installPlugin(Ogre::Plugin*)
#2    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/RenderSystem_GL3Plus.so", at 0x7fb678182c4d, in Ogre::GL3PlusPixelFormatToShaderType::getPixelFormatType(Ogre::PixelFormat) const
#1    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/RenderSystem_GL3Plus.so", at 0x7fb678189360, in Ogre::GL3PlusRenderSystem::GL3PlusRenderSystem()
#0    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/RenderSystem_GL3Plus.so", at 0x7fb678160115, in Ogre::GLSLShaderManager::unregisterShaderFactory(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
Segmentation fault (Invalid permissions for mapped object [0x7fb66c241818])
```

m3d, 2020-09-16:
p.s. it looks like I have other problems related to system upgrade (after a short blackout in Prague, sigh). I have problems also with other models, and what reported by my colleagues with Freyja config 2 is something else.

m3d, 2020-09-16:
OK, I tried ROBOTIKA_FREYJA_SENSOR_CONFIG_2 directly on CloudSim (robotika, ver69tbs3, 4727cbc9-6c8f-44b9-b98b-e8a3e0015ea8) and there are plenty of
```
[Dbg] [WebsocketServer.cc:260] LWS_CALLBACK_HTTP
[Dbg] [WebsocketServer.cc:144] Requested URI: /
[Dbg] [WebsocketServer.cc:189] Resource not found.
```

nkoenig, 2020-09-17:
Those are debug statements for the websocket server, which support 3D web visualization. They are fine.

m3d, 2020-09-17:
@nkoenig thanks - I will double check with my colleague what was/is exactly the issue ...

OT regarding segmentation fault (original report) it seems to be too similar to https://github.com/osrf/subt/issues/444 except I am using docker images ... and I have gcc version 7.5

adlarkin, 2021-01-15:
I just ran the following command with the latest `osrf/subt-virtual-testbed` image, and didn't experience a crash on simulator start:

```
./run.bash osrf/subt-virtual-testbed competition.ign robotName1:=X1 robotConfig1:=ROBOTIKA_FREYJA_SENSOR_CONFIG_2
```

So, I think that this has been resolved in one of the releases since `2020-09-01`. @m3d, can you confirm that you aren't seeing this issue anymore? If so, I would like to close this issue.

nkoenig, 2021-06-21:
Closing due to no additional comments.

closed 2021-06-21 by nkoenig