osrf/subt #613: Error Pool overlaps with json2docker.rb

knoedler, 2020-09-17:
I attempted to use json2docker.rb, but when I attempt to run the sample the output I see is:
./json2docker.rb -e > sample.json
./json2docker.rb -f sample.json -r
Creating network "tmp_sim_net" with the default driver
ERROR: Pool overlaps with other one on this address space
Removing network tmp_sim_net
WARNING: Network tmp_sim_net not found.
Removing network tmp_relay_net1
WARNING: Network tmp_relay_net1 not found.
Removing network tmp_relay_net2
WARNING: Network tmp_relay_net2 not found.

This was using subt pulled as of 2020_09_17 at 13:00

malcolmst, 2020-09-19:
I haven’t been using the new json2docker scripts yet, but I *think* I may have seen that error before when I had an existing network that was too similar to a new network. You could maybe try running `docker network prune` or manually delete any existing docker networks to see if that helps.

knoedler, 2020-09-21:
Thank you for the suggestion.   docker network prune   did get it working for me.

nkoenig, 2020-09-21:
Yup, docker sometimes requires cleaning using `docker network prune. I added this to the [wiki](https://github.com/osrf/subt/wiki/Cloudsim-Docker-Compose#error-pool-overlaps-with-other-one-on-this-address-space).

closed 2020-09-21 by nkoenig