osrf/subt #617: power_load parameter

pauljurczak, 2020-09-21:
Is increasing `power_load` parameter used exclusively to reduce battery life from 20 minutes for UAV or it affects performance of the robot, e.g. higher maximum speed with higher `power_load`?

azeey, 2020-09-21:
If the question is whether the simulator looks at the `power_load` parameter to determine the maximum speed, the answer is no. The maximum speed is set by the parameters of the `MulticopterVelocityControl` and `MulticopterMotorModel` plugins.

pauljurczak, 2020-09-21:
Yes, that was my question. In the real world you could sacrifice battery life for higher performance. I wanted to verify that this is not being the case in this simulation.

azeey, 2020-09-21:
That's true. Unfortunately, only constant power loads are supported in the simulator right now.

closed 2020-09-21 by azeey