osrf/subt #620: Playback model errors

rsawtell, 2020-09-23:
When I try to playback cloudsiim logs, I get a slew of errors along the lines of:

`[GUI] [Err] [SystemPaths.cc:459] File [/home/developer/.ignition/fuel/fuel.ignitionrobotics.org/OpenRobotics/models/Base Station/6/meshes/tent.dae] resolved to path [/home/developer/.ignition/fuel/fuel.ignitionrobotics.org/OpenRobotics/models/Base Station/6/meshes/tent.dae] but the path does not exist`

I looked for the file and found it in

`/home/developer/.ignition/fuel/fuel.ignitionrobotics.org/OpenRobotics/models/Base Station/1/meshes/tent.dae`

(see the 1 before meshes and after Base Station as opposed to a 6)

I'm also able to see these models perfectly fine when running local simulations, it's only a problem with the playback of logs.

nkoenig, 2020-09-24:
Those errors indicate that your don't have the correct model version downloaded locally. The log files were created with version 6 of the `Base Station` model, but you have version 1 on you local computer. 

You can download the latest version of the entire SubT Collection using:

```
ign fuel download -v 4 -j 16 -u "https://fuel.ignitionrobotics.org/OpenRobotics/collections/SubT Tech Repo"
```

If you want to download a specific model, then use:

```
ign fuel download -v 4 -j 16 -u "https://fuel.ignitionrobotics.org/OpenRobotics/models/{MODEL_NAME}"
```

rsawtell, 2020-09-25:
That fixed it, thanks

angelacmaio, 2020-09-26:
Resolved

closed 2020-09-26 by angelacmaio

peci1, 2020-09-26:
I hoped that this redownload would happen automatically...