osrf/subt #623: Problem with CMake build due to libignition-math6 version bump

pauljurczak, 2020-09-24:
CMake build, stopped working after upgrade to 2020-09-01 release. There was a version upgrade of `libignition-math6` from 6.5.0 to 6.6.0 and I'm getting this build error:
```
[build] make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libignition-math6.so.6.5.0', needed by '/home/paul/st/docker/libsimc.so'.  Stop.
```
because `libignition-math6.so.6.5.0` no longer exists. My `CMakeLists.txt` file:
```
cmake_minimum_required(VERSION 3.10.0)
project(simc VERSION 0.2.0 LANGUAGES CXX)

add_library(simc SHARED simc.cpp)

target_compile_options(simc PRIVATE -fPIC)

find_package(catkin REQUIRED COMPONENTS
  roscpp
  std_msgs
  rosbag
  tf
  genmsg
  laser_geometry
  message_generation
  subt_ign
)

include_directories(
  /home/paul/subt_ws/install/include
  /opt/ros/melodic/include
  /usr/include/ignition/transport7
  /usr/include/ignition/msgs4
  /usr/include/ignition/math6  
  ${catkin_INCLUDE_DIRS}
)

target_link_libraries(simc 
  ${catkin_LIBRARIES} 
)
```

nkoenig, 2020-09-24:
It doesn't look like the attached CMake file explicitly requires version 6.5.0 of ignition-math. I would expect this to compile, unless there is another CMake file in your build system that requires a specific minor version of ignition-math.

pauljurczak, 2020-09-24:
I reduced `CMakeLists.txt` to:
```
cmake_minimum_required(VERSION 3.10.0)
project(simc VERSION 0.2.0 LANGUAGES CXX)

add_library(simc SHARED simc.cpp)

target_compile_options(simc PRIVATE -fPIC)

find_package(catkin REQUIRED COMPONENTS
  roscpp
  std_msgs
  subt_ign
)

include_directories(
  /home/paul/subt_ws/install/include
  /opt/ros/melodic/include
  ${catkin_INCLUDE_DIRS}
)

target_link_libraries(simc 
  ${catkin_LIBRARIES} 
)
```
and there are no other dependencies than ROS and Ignition in this project. CMake configuration finds correct versions of dependencies:
```
[cmake] Looking for ignition-msgs4 -- found version 4.9.0
[cmake] Searching for dependencies of ignition-msgs4
[cmake] Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread;-lpthread (found suitable version "3.0.0", minimum required is "3") 
[cmake] Looking for ignition-math6 -- found version 6.6.0
[cmake] Searching for dependencies of ignition-math6
[cmake] Looking for ignition-transport7 -- found version 7.5.0
[cmake] Configuring done
[cmake] Generating done
```
but generated `link.txt` and `build.make` files still contain both old and new versions of `libignition-math6.so`. I tried `ldconfig`, but `ld.so.cache` contains the correct link to the newest version. I observed the same behavior on two different systems. This project was unchanged for a while and I didn't have this problem during previous SubT version upgrades.

pauljurczak, 2020-09-24:
Here is the content of `link.txt`, which for some reason has 3 copies of `libignition-math6.so`, 2 old and 1 new version:
```
/usr/bin/g++-8 -fPIC -O3 -DNDEBUG  -shared -Wl,-soname,libsimc.so -o /home/paul/st/docker/libsimc.so CMakeFiles/simc.dir/simc.cpp.o -Wl,-rpath,/home/paul/subt_ws/install/lib:/opt/ros/melodic/lib /usr/lib/x86_64-linux-gnu/libprotobuf.so -lpthread -lpthread /usr/lib/x86_64-linux-gnu/libprotobuf.so -lpthread -lpthread /home/paul/subt_ws/install/lib/libSubtProtobuf.so /home/paul/subt_ws/install/lib/libsubt_communication_broker.so /home/paul/subt_ws/install/lib/libsubt_communication_client.so /home/paul/subt_ws/install/lib/libSubtCommsProtobuf.so /opt/ros/melodic/lib/libroscpp.so /usr/lib/x86_64-linux-gnu/libboost_filesystem.so /opt/ros/melodic/lib/librosconsole.so /opt/ros/melodic/lib/librosconsole_log4cxx.so /opt/ros/melodic/lib/librosconsole_backend_interface.so /usr/lib/x86_64-linux-gnu/liblog4cxx.so /usr/lib/x86_64-linux-gnu/libboost_regex.so /opt/ros/melodic/lib/libxmlrpcpp.so /home/paul/subt_ws/install/lib/libsubt_communication_model.so /home/paul/subt_ws/install/lib/libsubt_rf_range_model.so /usr/lib/x86_64-linux-gnu/libignition-math6.so.6.5.0 /opt/ros/melodic/lib/libroscpp_serialization.so /opt/ros/melodic/lib/librostime.so /opt/ros/melodic/lib/libcpp_common.so /usr/lib/x86_64-linux-gnu/libboost_system.so /usr/lib/x86_64-linux-gnu/libboost_thread.so /usr/lib/x86_64-linux-gnu/libboost_chrono.so /usr/lib/x86_64-linux-gnu/libboost_date_time.so /usr/lib/x86_64-linux-gnu/libboost_atomic.so -lpthread /usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.4 /usr/lib/x86_64-linux-gnu/libignition-transport7.so.7.5.0 /usr/lib/x86_64-linux-gnu/libignition-msgs4.so.4.9.0 /usr/lib/x86_64-linux-gnu/libprotobuf.so /usr/lib/x86_64-linux-gnu/libignition-math6.so.6.6.0 /usr/lib/x86_64-linux-gnu/libuuid.so /usr/lib/x86_64-linux-gnu/libuuid.so /usr/lib/x86_64-linux-gnu/libprotobuf.so /usr/lib/x86_64-linux-gnu/libprotobuf.so -lpthread /home/paul/subt_ws/install/lib/libSubtProtobuf.so /home/paul/subt_ws/install/lib/libsubt_communication_broker.so /home/paul/subt_ws/install/lib/libsubt_communication_client.so /home/paul/subt_ws/install/lib/libSubtCommsProtobuf.so /opt/ros/melodic/lib/libroscpp.so /usr/lib/x86_64-linux-gnu/libboost_filesystem.so /opt/ros/melodic/lib/librosconsole.so /opt/ros/melodic/lib/librosconsole_log4cxx.so /opt/ros/melodic/lib/librosconsole_backend_interface.so /usr/lib/x86_64-linux-gnu/liblog4cxx.so /usr/lib/x86_64-linux-gnu/libboost_regex.so /opt/ros/melodic/lib/libxmlrpcpp.so /home/paul/subt_ws/install/lib/libsubt_communication_model.so /home/paul/subt_ws/install/lib/libsubt_rf_range_model.so /usr/lib/x86_64-linux-gnu/libignition-math6.so.6.5.0 /opt/ros/melodic/lib/libroscpp_serialization.so /opt/ros/melodic/lib/librostime.so /opt/ros/melodic/lib/libcpp_common.so /usr/lib/x86_64-linux-gnu/libboost_system.so /usr/lib/x86_64-linux-gnu/libboost_thread.so /usr/lib/x86_64-linux-gnu/libboost_chrono.so /usr/lib/x86_64-linux-gnu/libboost_date_time.so /usr/lib/x86_64-linux-gnu/libboost_atomic.so -lpthread /usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.4
```

nkoenig, 2020-09-25:
It looks like the `link.txt` file has both math 6.5.0 and 6.6.0. I think the 6.5.0 versions are referenced by subt code. There is no explicit reference to 6.5.0 in the subt code base. Do you have a clean environment and build?

I don't suppose you could share this project with me?

pauljurczak, 2020-09-25:
I found the problem. Specific version, `libignition-math6.so.6.4.0` on one system and `libignition-math6.so.6.5.0` on another, is hard coded in these files:
```
home/paul/subt_ws/install/lib/pkgconfig/subt_rf_interface.pc
home/paul/subt_ws/devel/lib/pkgconfig/subt_rf_interface.pc
home/paul/subt_ws/install/share/subt_rf_interface/cmake/subt_rf_interfaceConfig.cmake
home/paul/subt_ws/devel/share/subt_rf_interface/cmake/subt_rf_interfaceConfig.cmake
```
My projects build fine after I substituted hard coded version with symlink `libignition-math6.so.6` in all of these files. I don't know CMake well enough to judge, which of them were guilty.

pauljurczak, 2020-09-27:
This seems to be fixed in the master branch as of 09-26-2020.

closed 2020-09-27 by pauljurczak