osrf/subt #625: Extract all ROS messages into a message-only package

peci1, 2020-09-24:
After cave, I'd really suggest to do some refactoring and move all ROS message and service definitions to a message-only package, as is a good manner in other ROS projects. That would allow people to actually create solution Docker images which do not contain the Subt repo with all its large dependencies, but just with one dependency-less package. That would be heaven :-D

To make the refactoring really useful, it'd also be nice to separate the parts of submitted models that actually build ignition plugins into separate packages. That would allow people to pull the robot resources (meshes, urdf) without needing to install Ignition stuff.

zwn, 2020-09-24:
You would still need #138 for that to work. We tried to propose that already in #35 all the way back in 2018 :worried:. For some context see also #113.

peci1, 2020-09-25:
You're right, comms client would also drag in subt workspace dependencies...