osrf/subt #627: Feature suggestion: show score in simulation summary and email summary

malcolmst, 2020-09-26:
Not sure if this might already be on your radar, but just wanted to suggest that it would be helpful to have the score of each simulation shown in the simulation summary and the email that gets sent out on completion. That would help to better determine which simulations to look at for a more detailed analysis (i.e. if it did very well or very poorly) without having to download the logs first.

realdealneil, 2020-10-13:
I agree with this one.  It would be really nice to see the score on the web summary page.

zwn, 2020-10-13:
We have been asking for such a thing for some time #223 already. From the title I would say it was asked during Tunnel and moved to Urban and then forgotten. Maybe it is a minor enhancement from the implementation point of view but for sure it is a major enhancement for our productivity.

nkoenig, 2021-01-14:
Score information should now appear on Portal, and summary emails will contain the `run.yml` file (https://gitlab.com/ignitionrobotics/web/cloudsim/-/merge_requests/217).

closed 2021-01-14 by nkoenig

peci1, 2021-01-14:
Thanks for hearing us! The score in cloudsim is indeed really practical.

Now, it'd be nice if we received the simulation log emails again, as they stopped coming to me around 16 Oct 2020 (and I'm not the only one from our group).

realdealneil, 2021-01-14:
Excellent!  Thanks for implementing this!