osrf/subt #628: No ground truth pose on /world topic

pauljurczak, 2020-09-26:
I posted to a closed issue #325 and I'm not sure what is the proper etiquette in such case, but I would like to get an answer.

I just reinstalled SubT from master and I don't see `/world` topic
```
paul@desktop:~$ ign launch -v 0 cave_circuit.ign worldName:=cave_qual robotName1:=X1 robotConfig1:=X3_SENSOR_CONFIG_5 enableGroundTruth:=true
...
paul@desktop:~$ rostopic list
/X1/air_pressure
/X1/battery_state
/X1/bottom_scan
/X1/cmd_vel
/X1/front/camera_info
/X1/front/depth
/X1/front/depth/compressed
/X1/front/depth/compressed/parameter_descriptions
/X1/front/depth/compressed/parameter_updates
/X1/front/depth/compressedDepth
/X1/front/depth/compressedDepth/parameter_descriptions
/X1/front/depth/compressedDepth/parameter_updates
/X1/front/depth/theora
/X1/front/depth/theora/parameter_descriptions
/X1/front/depth/theora/parameter_updates
/X1/front/image_raw
/X1/front/image_raw/compressed
/X1/front/image_raw/compressed/parameter_descriptions
/X1/front/image_raw/compressed/parameter_updates
/X1/front/image_raw/compressedDepth
/X1/front/image_raw/compressedDepth/parameter_descriptions
/X1/front/image_raw/compressedDepth/parameter_updates
/X1/front/image_raw/theora
/X1/front/image_raw/theora/parameter_descriptions
/X1/front/image_raw/theora/parameter_updates
/X1/front/optical/camera_info
/X1/front/optical/depth
/X1/front/optical/image_raw
/X1/gas_detected
/X1/imu/data
/X1/magnetic_field
/X1/output/camera_info
/X1/pose
/X1/pose_static
/X1/rgbd_camera/depth/points
/X1/top_scan
/X1/velocity_controller/enable
/clock
/rosout
/rosout_agg
/statistics
/subt/run_clock
/subt/score
/tf
/tf_static
```

peci1, 2020-09-26:
I've never seen a /world ROS topic. I've seen `world` TF frame, though, and `/world` namespace for Ignition topics (which would get listed via `ign topic -l`).

pauljurczak, 2020-09-26:
Thank you. I didn't know there are Ignition topics alongside of ROS topics. Can I subscribe to them using ROS API, i.e. `ros::Subscriber`, or there is a separate Ignition API for that? Are Ignition SubT topics described somewhere?

peci1, 2020-09-26:
ROS and Ignition are two completely separate worlds joined by https://github.com/ignitionrobotics/ros_ign/tree/melodic/ros_ign_bridge . You can look there to get inspiration. Ignition is used inside the simulator and ROS is an "outside" communication channel.

I think the idea for SubT is not to dive into Ignition that much, beacuse in the solution container you only have access to ROS. Ignition topics are not made available during the "final" evaluation of the submissions. But many of the tutorials like https://github.com/osrf/subt/wiki/Breadcrumbs-and-communication-visualization-tutorial use ignition topics/services to provide you insight into what's happening inside the simulator.

pauljurczak, 2020-09-27:
Great! That's what I wanted to know. I can use Ignition topics for development and testing in catkin builds, but they are not available in cloudsim, right?

peci1, 2020-09-27:
That's it ;)

closed 2020-09-27 by pauljurczak