osrf/subt #629: Unauthorized request on subtchallenge.com/compete

peci1, 2020-09-26:
I've submitted a few simulations lately via the CLI and when I want to see their detail on the competition web page, I get Unauthorized error.

The simulation API returns 401 error with the following content:

```
{"errcode":4002,"msg":"Unauthorized request","extra":null,"errid":"ea8c3208-fe7a-4330-9c87-9457b62cce45","route":"GET /1.0/simulations/8f383039-bcc2-4d04-b860-3d0fefca2336","user-agent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:81.0) Gecko/20100101 Firefox/81.0","remote-address":"90.181.31.138"}
```

When I access details about our earlier simulations from the same page, that works.

This is how I submitted the simulation:

```
$ curl -ks -X POST --url https://cloudsim.ignitionrobotics.org/1.0/simulations -F name=tradrTest2 -F owner=ctu-cras-norlab -F 'circuit=Cave Qualification' -F robot_name=X1 -F robot_type=CTU_CRAS_NORLAB_ABSOLEM_SENSOR_CONFIG_2 -F robot_image=200670743174.dkr.ecr.us-east-1.amazonaws.com/subt/ctu-cras-norlab:X1 -F robot_name=X2 -F robot_type=CTU_CRAS_NORLAB_ABSOLEM_SENSOR_CONFIG_2 -F robot_image=200670743174.dkr.ecr.us-east-1.amazonaws.com/subt/ctu-cras-norlab:X2 -F robot_name=TEAMBASE -F robot_type=TEAMBASE -F robot_image=200670743174.dkr.ecr.us-east-1.amazonaws.com/subt/ctu-cras-norlab:TEAMBASE --header 'Private-Token: ---'
{"created_at":"2020-09-26T11:56:35.220925075Z","updated_at":"2020-09-26T11:56:35.220925075Z","valid_for":"48h0m0s","owner":"ctu-cras-norlab","creator":"peci1","private":true,"stop_on_end":false,"name":"tradrTest2","image":"","group_id":"8f383039-bcc2-4d04-b860-3d0fefca2336","parent":null,"MultiSim":0,"status":0,"platform":"subt","application":"subt","extra":"{\"circuit\":\"Cave Qualification\",\"robots\":[{\"Name\":\"X1\",\"Type\":\"CTU_CRAS_NORLAB_ABSOLEM_SENSOR_CONFIG_2\",\"Image\":\"200670743174.dkr.ecr.us-east-1.amazonaws.com/subt/ctu-cras-norlab:X1\",\"Credits\":180},{\"Name\":\"X2\",\"Type\":\"CTU_CRAS_NORLAB_ABSOLEM_SENSOR_CONFIG_2\",\"Image\":\"200670743174.dkr.ecr.us-east-1.amazonaws.com/subt/ctu-cras-norlab:X2\",\"Credits\":180},{\"Name\":\"TEAMBASE\",\"Type\":\"TEAMBASE\",\"Image\":\"200670743174.dkr.ecr.us-east-1.amazonaws.com/subt/ctu-cras-norlab:TEAMBASE\",\"Credits\":0}]}","robots":"X1,X2,TEAMBASE","held":false}
```

nkoenig, 2020-09-29:
Hi @peci1, is this still an issue for you? We checked a few things on our side, and could see an obvious error. This might have occurred when we updated the production servers.

peci1, 2020-09-29:
I still receive the unauthorized error for group ids:

43456ed6-6a04-42e6-9b46-8cab356c69ce
4f2a06ad-85f7-4a57-9fa9-debd58cfcbb2
6089cc1d-379e-4307-8bcb-92f1c2dc96d4
c1bad22a-cae9-4fb1-a783-7ffad2bfcece
8f383039-bcc2-4d04-b860-3d0fefca2336

peci1, 2020-09-29:
It is possible that some of these submissions were kind of invalid, e.g. pointing to non-existing docker images etc.

zwn, 2020-10-01:
I experience the same problem now:
- f0b64ffe-0502-4e56-aafa-0ce230cc2869
- 18437fc9-a228-4e6e-901c-0a3f7f5c5814

peci1, 2020-10-01:
Please, give this a higher priority now that we know that I'm not the single one who experiences these issues... I've now submitted new simulations via web interface, both failed with `Terminated Error: InitializationFailed` and logs are inaccessible... We need this for debugging...

For these simulations, it is not UnauthorizedError, but Unkown error.

This is the request for logfiles download:

```
$ wget "https://web-cloudsim-production-logs.s3.us-west-2.amazonaws.com/gz-logs/CTU-CRAS-NORLAB/ac5683cd-6ce9-44a8-9396-d64faf6136e9/ac5683cd-6ce9-44a8-9396-d64faf6136e9-fc-teambase-commsbridge.tar.gz?X-Amz-Algorithm=AWS4-HMAC-SHA256&amp;X-Amz-Credential=XXX&amp;X-Amz-Date=20201001T132736Z&amp;X-Amz-Expires=300&amp;X-Amz-SignedHeaders=host&amp;X-Amz-Signature=XXX"
The name is too long, 353 chars total.
Trying to shorten...
New name is ac5683cd-6ce9-44a8-9396-d64faf6136e9-fc-teambase-commsbridge.tar.gz?X-Amz-Algorithm=AWS4-HMAC-SHA256&amp;X-Amz-Credential=XXXamp;X-Amz-Date=20201001T132736Z&amp;X-Amz-Expires=.
--2020-10-01 15:31:05--  https://web-cloudsim-production-logs.s3.us-west-2.amazonaws.com/gz-logs/CTU-CRAS-NORLAB/ac5683cd-6ce9-44a8-9396-d64faf6136e9/ac5683cd-6ce9-44a8-9396-d64faf6136e9-fc-teambase-commsbridge.tar.gz?X-Amz-Algorithm=AWS4-HMAC-SHA256&amp;X-Amz-Credential=XXX%2Faws4_request&amp;X-Amz-Date=20201001T132736Z&amp;X-Amz-Expires=300&amp;X-Amz-SignedHeaders=host&amp;X-Amz-Signature=XXX
Resolving web-cloudsim-production-logs.s3.us-west-2.amazonaws.com (web-cloudsim-production-logs.s3.us-west-2.amazonaws.com)... 52.218.136.161
Connecting to web-cloudsim-production-logs.s3.us-west-2.amazonaws.com (web-cloudsim-production-logs.s3.us-west-2.amazonaws.com)|52.218.136.161|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: unspecified
ERROR: Redirection (301) without location.
```

zwn, 2020-10-01:
When I click "Get Logs" I see this in the console
```
GEThttps://cloudsim.ignitionrobotics.org/1.0/simulations/f0b64ffe-0502-4e56-aafa-0ce230cc2869/logs/file?link=true 401 Unauthorized
```
response body
```
{"errcode":4002,"msg":"Unauthorized request","extra":null,"errid":"4a435d45-baee-462f-a547-70eccc405e80","route":"GET /1.0/simulations/f0b64ffe-0502-4e56-aafa-0ce230cc2869/logs/file?link=true","user-agent":"Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:81.0) Gecko/20100101 Firefox/81.0","remote-address":"77.240.103.204"}
```
Let me know if I can help with debugging in any way. Not being able to download logs is a real show stopper :worried:.

m3d, 2020-10-01:
I have the same problem as @zwn although I was able to download 18437fc9-a228-4e6e-901c-0a3f7f5c5814 files at 12:07 (now is 15:50), ie. 3.5 hours ago. Also by accident(?) one file was downloaded 3 times (and was blocked by browsed for multiple download). Now I am not able do download anything even after logout/login. (alternative is "Unknown error")

nkoenig, 2020-10-01:
Downloading logs should be fixed now. Give it a try, and post back if you run into problems.

peci1, 2020-10-01:
Thanks. The originally reported Unauthorized access errors are not fixed, but the later Unkown error problems are gone.

zwn, 2020-10-01:
Thanks. The logs are downloading now.

peci1, 2020-10-08:
I've found the culprit for the UnauthorizedError. I've submitted with owner="ctu-cras-norlab", whereas it should have been CTU-CRAS-NORLAB. I didn't consider the option that cloudsim would accept a simulation with a wrong owner.

nkoenig, 2020-10-08:
Organization name is case insensitive. @peci1 , in this instances there was a problem with the solution docker image.

peci1, 2020-10-08:
That's not in line with my observations. I know the docker image had problems, but this has nothing to do with this issue.

I've created 3 submissions:

- via webpage: 86a179f6-74f1-40c0-baa6-298f2ffaef61 - no problem, I can access the logs
- via REST API with lowercase owner: 53cb478e-4eea-4233-9821-6506de3beca7 - Error 4002 Unauthorized request when accessing logs
- via REST API with uppercase owner: 1a2d9cff-d838-4980-93c1-2a8d49510180 - no problem

To me, it looks clearly like a problem created by submitting the lowercase owner.

zwn, 2020-10-09:
We submit also with lower case owner and everything is working as expected.

peci1, 2021-01-19:
I'll close this issue, because it seems it was caused by some unrelated problem in cloudsim (solution containers not starting?).

closed 2021-01-19 by peci1