osrf/subt #630: Depth image noise

pauljurczak, 2020-09-27:
Until a while ago, there was no noise added to depth images other than some small 3D geometry artifacts. Recently, I noticed a significant noise add to depth data from VGA RGBD camera in X3 and DS1 robots. Is this an intentional improvement in simulation realism across all RGBD and ToF cameras?

angelacmaio, 2020-09-28:
Support for RGBD noise was added with: https://github.com/ignitionrobotics/ign-rendering/pull/122
As a result, the noise parameters that have always resided in this repo became supported for improved realism.

closed 2020-09-28 by angelacmaio

zwn, 2020-09-28:
It would have been nice if such a major change was mentioned at https://github.com/osrf/subt/wiki/release_notes