osrf/subt #631: Cloudsim crash in libGameLogicPlugin.so: Segmentation fault (Address not mapped to object [0xffffffffffffffc8])

malcolmst, 2020-09-27:
Just saw this crash in the logs for simulation ID 709794dc-eab6-4dec-be67-62b1e7a5d3fe (which is still running at the moment, but I can't connect to it for the active run visualization):

```
Stack trace (most recent call last) in thread 239:
#10   Object "", at 0xffffffffffffffff, in 
#9    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fa453c7fa3e, in clone
#8    Object "/lib/x86_64-linux-gnu/libpthread.so.0", at 0x7fa4539466da, in start_thread
#7    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fa450be66de, in std::error_code::default_error_condition() const
#6    Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo2.so.2", at 0x7fa448b0934c, in ignition::gazebo::v2::SimulationRunner::UpdateCurrentInfo()
#5    Object "/home/developer/subt_ws/install/lib/libGameLogicPlugin.so", at 0x7fa40ba8c508, in subt::GameLogicPlugin::PostUpdate(ignition::gazebo::v2::UpdateInfo const&, ignition::gazebo::v2::EntityComponentManager const&)
#4    Object "/home/developer/subt_ws/install/lib/libGameLogicPlugin.so", at 0x7fa40baab77c, in void ignition::gazebo::v2::EntityComponentManager::Each<ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > >, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > >(ignition::gazebo::v2::EntityComponentManager::identity<std::function<bool (unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*)> >::type) const
#3    Object "/home/developer/subt_ws/install/lib/libGameLogicPlugin.so", at 0x7fa40bac1423, in std::function<bool (unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*)>::operator()(unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*) const
#2    Source "GameLogicPlugin.cc", line 0, in std::_Function_handler<bool (unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*), subt::GameLogicPlugin::PostUpdate(ignition::gazebo::v2::UpdateInfo const&, ignition::gazebo::v2::EntityComponentManager const&)::{lambda(unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*)#4}>::_M_invoke(std::_Any_data const&, unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*&&, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*&&) [0x7fa40ba96502]
#1    Source "GameLogicPlugin.cc", line 0, in subt::GameLogicPlugin::PostUpdate(ignition::gazebo::v2::UpdateInfo const&, ignition::gazebo::v2::EntityComponentManager const&)::{lambda(unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*)#4}::operator()(unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*) const [0x7fa40ba8acdb]
#0    Object "/home/developer/subt_ws/install/lib/libGameLogicPlugin.so", at 0x7fa40baaad96, in ignition::math::v6::Vector3<double>::Distance(ignition::math::v6::Vector3<double> const&) const
Segmentation fault (Address not mapped to object [0xffffffffffffffc8])
./run_sim.bash: line 16:    54 Segmentation fault      (core dumped) ign launch -v 4 $@
```

I haven't seen this before, but also haven't been looking for it. I'll update if I find out anything else.

zwn, 2020-09-29:
I believe I might have the same crash (locally):

```
Stack trace (most recent call last) in thread 188:
#10   Object "", at 0xffffffffffffffff, in 
[Dbg] [VisibilityRfModel.cc:80] Range: 0, Exp: 2.5, TX: 20, RX: inf
[Dbg] [VisibilityRfModel.cc:80] Range: 0, Exp: 2.5, TX: 20, RX: inf
#9    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fdaf02a2a3e, in clone
#8    Object "/lib/x86_64-linux-gnu/libpthread.so.0", at 0x7fdaeff696da, in start_thread
#7    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fdaed2096de, in std::error_code::default_error_condition() const
#6    Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo2.so.2", at 0x7fdad3d6e71c, in ignition::gazebo::v2::SimulationRunner::UpdateCurrentInfo()
#5    Object "/home/developer/subt_ws/install/lib/libGameLogicPlugin.so", at 0x7fdaaf47662b, in subt::GameLogicPlugin::PostUpdate(ignition::gazebo::v2::UpdateInfo const&, ignition::gazebo::v2::EntityComponentManager const&)
#4    Object "/home/developer/subt_ws/install/lib/libGameLogicPlugin.so", at 0x7fdaaf490e54, in void ignition::gazebo::v2::EntityComponentManager::Each<ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > >, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > >(ignition::gazebo::v2::EntityComponentManager::identity<std::function<bool (unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*)> >::type) const
#3    Object "/home/developer/subt_ws/install/lib/libGameLogicPlugin.so", at 0x7fdaaf49ded5, in std::function<bool (unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*)>::operator()(unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*) const
#2    Source "GameLogicPlugin.cc", line 0, in std::_Function_handler<bool (unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*), subt::GameLogicPlugin::PostUpdate(ignition::gazebo::v2::UpdateInfo const&, ignition::gazebo::v2::EntityComponentManager const&)::{lambda(unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*)#4}>::_M_invoke(std::_Any_data const&, unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*&&, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*&&) [0x7fdaaf47ed4e]
#1    Source "GameLogicPlugin.cc", line 0, in subt::GameLogicPlugin::PostUpdate(ignition::gazebo::v2::UpdateInfo const&, ignition::gazebo::v2::EntityComponentManager const&)::{lambda(unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*)#4}::operator()(unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, igni [... cut ...] on::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*) const [0x7fdaaf47554f]
#0    Object "/home/developer/subt_ws/install/lib/libGameLogicPlugin.so", at 0x7fdaaf490318, in ignition::math::v6::Vector3<double>::Distance(ignition::math::v6::Vector3<double> const&) const
Segmentation fault (Address not mapped to object [0xffffffffffffffe8])
./run_sim.bash: line 16:    54 Segmentation fault      (core dumped) ign launch -v 4 $@
```

zwn, 2020-09-29:
I am able to reproduce this crash in docker release 2020-09-01 and 2020-09-25. Currently, I am not able to run anything :worried:

zwn, 2020-09-29:
Ok, so on my side, the crash seems to have been caused by missing `circuit:=cave` for running cave worlds. It would be a nice improvement not to have this argument.

nkoenig, 2020-09-29:
Do you happen to have some more information about when during a simulation this crash occurs? For example, does it happen when you drop a breadcrumb, or report an artifact?

Also, could you send me (nate@osrfoundation.org or via subt-help@googlegroups.com) the run configuration that you use to reproduce this locally?

zwn, 2020-09-29:
@nkoenig If that was aimed at me, then the problem appears when I call the service to get robot position wrt artifact origin. It either crashes or reports that the robot is out of range (and crashes on the next call). This happens for any robot in any **cave** world when there is `circuit:=urban` on the command line to start the simulation docker (as was forgotten in one of my scripts). 

Since calling that service is the first thing I did, it crashed immediately.

nkoenig, 2020-09-29:
Got it, thanks for the info.

malcolmst, 2020-09-29:
Mine may very well have likely crashed during an artifact origin call too. For me though it’s not a consistent crash.

nkoenig, 2020-09-30:
Would anyone be willing to try out the [osrf/subt-virtual-testbed:cloudsim_sim_issue_631](https://hub.docker.com/layers/osrf/subt-virtual-testbed/cloudsim_sim_issue_631/images/sha256-3aab563b61b08f1b5ba6610d476e75c1bb3c72ba9eca5e5dcdfe9b50f81884a5?context=repo) docker image? It's built off of [this branch](https://github.com/osrf/subt/compare/distance_segfault).

zwn, 2020-10-02:
I can confirm that with the supplier docker image I no longer get the crash I described at https://github.com/osrf/subt/issues/631#issuecomment-700800506.

zwn, 2020-10-05:
Can we deploy this ASAP? It seems we are hitting this on cloudsim. Simulation e614ff1c-15ba-475a-b423-77d47d603082 should have ran for 2300 simulated seconds but ran only 637. The last message I could find in rosout of one of the robots is:

```
710.012000000 ERROR /C900W600L_proxy [/osgar-ws/src/osgar/subt/ros/proxy/ros_proxy_node.cc:402(Update)] [topics: /rosout, /C900W600L/cmd_vel_osgar, /statistics] Failed to call pose_from_artifact_origin service, robot may not exist, be outside staging area, or the service is not available.
```

But this robot has not yet even started moving.

zwn, 2020-10-05:
Another one 84ffbecd-eaf5-44e1-8769-d2cd7a77c2f2-r-1:

```
#9    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f232e981a3e, in clone
#8    Object "/lib/x86_64-linux-gnu/libpthread.so.0", at 0x7f232e6486da, in start_thread
#7    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f232b8e86de, in std::error_code::default_error_condition() const
#6    Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo2.so.2", at 0x7f231f56334c, in ignition::gazebo::v2::SimulationRunner::UpdateCurrentInfo()
#5    Object "/home/developer/subt_ws/install/lib/libGameLogicPlugin.so", at 0x7f22ea59e508, in subt::GameLogicPlugin::PostUpdate(ignition::gazebo::v2::UpdateInfo const&, ignition::gazebo::v2::EntityComponentManager const&)
#4    Object "/home/developer/subt_ws/install/lib/libGameLogicPlugin.so", at 0x7f22ea5bd77c, in void ignition::gazebo::v2::EntityComponentManager::Each<ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > >, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > >(ignition::gazebo::v2::EntityComponentManager::identity<std::function<bool (unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*)> >::type) const
#3    Object "/home/developer/subt_ws/install/lib/libGameLogicPlugin.so", at 0x7f22ea5d3423, in std::function<bool (unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*)>::operator()(unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*) const
#2    Source "GameLogicPlugin.cc", line 0, in std::_Function_handler<bool (unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*), subt::GameLogicPlugin::PostUpdate(ignition::gazebo::v2::UpdateInfo const&, ignition::gazebo::v2::EntityComponentManager const&)::{lambda(unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*)#4}>::_M_invoke(std::_Any_data const&, unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*&&, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*&&) [0x7f22ea5a8502]
#1    Source "GameLogicPlugin.cc", line 0, in subt::GameLogicPlugin::PostUpdate(ignition::gazebo::v2::UpdateInfo const&, ignition::gazebo::v2::EntityComponentManager const&)::{lambda(unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*)#4}::operator()(unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*) const [0x7f22ea59ccdb]
#0    Object "/home/developer/subt_ws/install/lib/libGameLogicPlugin.so", at 0x7f22ea5bcd96, in ignition::math::v6::Vector3<double>::Distance(ignition::math::v6::Vector3<double> const&) const
Segmentation fault (Address not mapped to object [0xffffffffffffffc8])
./run_sim.bash: line 16:    54 Segmentation fault      (core dumped) ign launch -v 4 $@
```

The WebUI says it is still running so it is most likely still blocking cloudsim resources for all teams :worried:

malcolmst, 2020-10-05:
I have one as well in that state: e3a3f686-0bf4-4315-9877-789977f5026d-r-1

My other one running seems to be working fine though currently.

zwn, 2020-10-06:
> Can we deploy this ASAP?

What is holding us from deploying?

closed 2020-10-06 by angelacmaio

ala28, 2020-10-07:
FWIW, my previous 7 runs all seg faulted but the most recent using osrf/subt-virtual-testbed:cloudsim_sim_issue_631 finished successfully.

m3d, 2020-10-07:
Is it really already deployed on CloudSim? The simulation which was previously crashing is still crashing:

[image]

zwn, 2020-10-07:
There is no way to check the version running on cloudsim from competitors side :worried: 

That said, I don't see anyone mentioning making a release or deploying to cloudsim. The issue was closed automatically due to PR being merged. I don't see a way for me to reopen the issue to track this all the way to release and deployment. So let's just hope we will be notified anyway.

BTW: the last crashed simulation is 0a85169d-d7ac-4836-a76b-38762a81f8c1.

@angelacmaio @nkoenig Please let us know when a release is made and the fix is deployed to cloudsim. We are all holding our collective breaths until then.

malcolmst, 2020-10-07:
Just ran into another one too, unfortunately: `5c4dfd07-6827-4af2-9a70-b72977bb49b2`

nkoenig, 2020-10-08:
This has been released on Cloudsim, public dockerhub images are uploading. Release notes are here: https://github.com/osrf/subt/wiki/release_notes#2020-10-07,

zwn, 2020-10-09:
I can confirm that one of our strategies that have been consistently crashing due to this problem no longer crashes and runs to a successful end (on current cloudsim). Thanks for the fix!

ala28, 2020-10-11:
The seg fault is back. Not sure if it's the same or not...

```
#9    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f5850455a3e, in clone
#8    Object "/lib/x86_64-linux-gnu/libpthread.so.0", at 0x7f585011c6da, in start_thread
#7    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f584d3bc6de, in std::error_code::default_error_condition() const
#6    Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo2.so.2", at 0x7f584111134c, in ignition::gazebo::v2::SimulationRunner::UpdateCurrentInfo()
#5    Object "/home/developer/subt_ws/install/lib/libGameLogicPlugin.so", at 0x7f580c2e79aa, in subt::GameLogicPlugin::PostUpdate(ignition::gazebo::v2::UpdateInfo const&, ignition::gazebo::v2::EntityComponentManager const&)
#4    Object "/home/developer/subt_ws/install/lib/libGameLogicPlugin.so", at 0x7f580c309e7c, in void ignition::gazebo::v2::EntityComponentManager::Each<ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > >, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > >(ignition::gazebo::v2::EntityComponentManager::identity<std::function<bool (unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*)> >::type) const
#3    Object "/home/developer/subt_ws/install/lib/libGameLogicPlugin.so", at 0x7f580c32038d, in std::function<bool (unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*)>::operator()(unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*) const
#2    Source "GameLogicPlugin.cc", line 0, in std::_Function_handler<bool (unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*), subt::GameLogicPlugin::PostUpdate(ignition::gazebo::v2::UpdateInfo const&, ignition::gazebo::v2::EntityComponentManager const&)::{lambda(unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*)#4}>::_M_invoke(std::_Any_data const&, unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*&&, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*&&) [0x7f580c2f1e63]
#1    Source "GameLogicPlugin.cc", line 0, in subt::GameLogicPlugin::PostUpdate(ignition::gazebo::v2::UpdateInfo const&, ignition::gazebo::v2::EntityComponentManager const&)::{lambda(unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*)#4}::operator()(unsigned long const&, ignition::gazebo::v2::components::Component<std::add_lvalue_reference<void>, ignition::gazebo::v2::components::SensorTag, ignition::gazebo::v2::serializers::DefaultSerializer<std::add_lvalue_reference<void> > > const*, ignition::gazebo::v2::components::Component<unsigned long, ignition::gazebo::v2::components::ParentEntityTag, ignition::gazebo::v2::serializers::DefaultSerializer<unsigned long> > const*) const [0x7f580c2e6171]
#0    Object "/home/developer/subt_ws/install/lib/libGameLogicPlugin.so", at 0x7f580c309496, in ignition::math::v6::Vector3<double>::Distance(ignition::math::v6::Vector3<double> const&) const
Segmentation fault (Address not mapped to object [0xffffffffffffffc8])
./run_sim.bash: line 17:    54 Segmentation fault      (core dumped) ign launch -v 4 $@
[Dbg] [VisibilityRfModel.cc:134] Range: 92.9308
```

zwn, 2020-10-12:
If the stacktrace is from cloudsim, please add ID of the run. If it is from a local run, can you make sure you have the latest dockers?  The respective sha256 digests are on the release page. If you have the latest dockers, are you able to reproduce the crash? If so, any hints to as what the reproduction steps could be for someone else?

Thanks. [just trying to add my 2c]

ala28, 2020-10-12:
All on cloudsim:

e2556461-d80d-433f-919c-2beca0669f83
a38f161a-8086-494c-9cbd-b804a52e79d9
a6d01ca0-5ebf-412e-8543-a8e5ded3524e
etc...

I don't know about reproducing the crash but I think the last five all did crash the same.

PR #642 (2020-10-06): Distance segfault
This PR attempts to resolve #631 and fixes ROS bag recording on the simulation server. A docker image based on this is available using [osrf/subt-virtual-testbed:cloudsim_sim_issue_631](https://hub.docker.com/layers/osrf/subt-virtual-testbed/cloudsim_sim_issue_631/images/sha256-3aab563b61b08f1b5ba6610d476e75c1bb3c72ba9eca5e5dcdfe9b50f81884a5?context=repo).