osrf/subt #634: build problem with subt

wolfgangschwab, 2020-09-29:
The build of the subt package results in an error on my system.  I just downloaded the newest changes (branch: master). Now the build does not work any more. 

> In file included from /home/wolfgang/subt_ws/src/subt/subt_ign/src/validate_visibility_table.cc:19:0:
> /home/wolfgang/subt_ws/src/subt/subt_ign/include/subt_ign/VisibilityTable.hh:97:42: error: ‘shared_ptr’ is not a member of ‘std’
>          const std::map<std::string, std::shared_ptr<fcl::CollisionObject>> &_collObjs);
>                                           ^~~~~~~~~~
> /home/wolfgang/subt_ws/src/subt/subt_ign/include/subt_ign/VisibilityTable.hh:97:58: error: template argument 2 is invalid
>          const std::map<std::string, std::shared_ptr<fcl::CollisionObject>> &_collObjs);
>                                                           ^~~~~~~~~~~~~~~
> /home/wolfgang/subt_ws/src/subt/subt_ign/include/subt_ign/VisibilityTable.hh:97:58: error: template argument 4 is invalid
> /home/wolfgang/subt_ws/src/subt/subt_ign/include/subt_ign/VisibilityTable.hh:97:42: error: ‘shared_ptr’ is not a member of ‘std’
>          const std::map<std::string, std::shared_ptr<fcl::CollisionObject>> &_collObjs);
>                                           ^~~~~~~~~~
> ...

Any hint to solve this is appreciated.

rsawtell, 2020-09-29:
We've occasionally run into problems like this and they are usually solved by updating packages to the latest version and/or in the case of a local installation deleting the devel, build, and install folders in the catkin workspace and recompiling.

angelacmaio, 2020-09-29:
@wolfgangschwab have you tried running the update steps [here](https://github.com/osrf/subt/wiki/Catkin-System-Setup#updating-the-subt-simulator)?

nkoenig, 2020-09-29:
I'm not able to reproduce this problem, but maybe pull request #636 will resolve this issue. Can you give that a shot?

wolfgangschwab, 2020-09-29:
The pull request #636 solved the issue. 
Thanks Nate!

closed 2020-09-29 by wolfgangschwab

nkoenig, 2020-10-08:
This has been released on Cloudsim, public dockerhub images are uploading. Release notes are here: https://github.com/osrf/subt/wiki/release_notes#2020-10-07,

PR #636 (2020-09-30): Added memory include
Potentially resolves #634.