osrf/subt #635: Camera Rendering Broken on AWS

rsawtell, 2020-09-29:
Images we bagged from our latest run reveal some sort of camera simulation failure on the AWS, namely that all or part of the world appears to disappear from the perspective of the camera system leaving a mostly black screen in which only certain things (artifacts and other robots) appear to be rendered. The failure seems intermittent, but affected every robot at some point.

Valid image taken by a UAV, and a corrupted image a few seconds later:
[image]

Same UAV later in the simulation, part of the world is visible while other parts are not:
[image]

View of a rescue randy by a UGV, note the shadowed area at the bottom should be obscured by terrain (you may have to adjust your contrast but you can see the entire rescue randy, it's just darker):
[image]

Not sure if this is related or not, but the log playback seems mostly fine except depending on the perspective you get things like this (the camera is not inside any geometry):
[image]

or as a GIF:
[image]

For reference the simulation id is 165fd66a-ab7f-471e-91b7-a49cf616f129

nkoenig, 2020-09-29:
Thanks. I'm looking into it.

iche033, 2020-09-29:
@rsawtell, are the images from an rgbd camera? Do you know the approximate locations in the world where the UAV captured these images?

rsawtell, 2020-09-29:
These are from HD camera configurations. The locations appear to be almost anywhere in the tunnel, it's not really limited to specific locations. In the above examples they are the helmet to the right after entering the tunnel, the starting area, and the first rescue randy on the left branch of the tunnel. Probably 90% of the images I captured had the problem though from basically anywhere the robots visited, and this seems new to this run as I've been capturing images for awhile now. I also don't see it in local development testing, though i haven't tried cloudsim.

peci1, 2020-09-30:
Just to make sure - do you use an english locale in your system? Try running the GUI with environment `LC_ALL=C LANG=C`.

rsawtell, 2020-09-30:
We are running on an english locale.

We also saw this problem on our latest run: 
Run ID: b964c8b7-b8a5-42c1-9f31-b1368ed2be60

[image]

nkoenig, 2020-10-01:
The issue is caused by surpassing a shadow caster limit in Ogre2. A patch is in progress. I the mean time you have a couple options:

1. Run with fewer robots. I realize this is not a long term solution.
2. Turn off shadow casting on some of the lights on your robots. This solution will only affect local runs. Look for the `<cast_shadows>` tag in the `model.sdf` file.

I'll update this issue when the patch is in place.

rsawtell, 2020-10-02:
Thanks for the update, we'll keep this in mind.

malcolmst, 2020-10-07:
For robot budgeting purposes, is it safe to assume this will be fixed before cave circuit?

nkoenig, 2020-10-08:
Fix has been on Cloudsim, public dockerhub images are uploading. Release notes are here: https://github.com/osrf/subt/wiki/release_notes#2020-10-07,

closed 2020-10-08 by nkoenig

malcolmst, 2020-10-08:
Awesome! Thanks for the update.