osrf/subt #639: Simulation Ended Shortly After Starting

rsawtell, 2020-09-30:
In our latest run, the entire simulation ended early, listing the total simulation time as 122 seconds. Our robot logs seem to indicate it went a bit longer, to approximately 181 seconds, after which the clock topic stopped publishing but the container kept running, which continued from 2020-09-29 15:17:54 to 2020-09-30 06:56:55 after which the logs stop.

Run ID: b964c8b7-b8a5-42c1-9f31-b1368ed2be60

rsawtell, 2020-10-01:
Happened again for me with Run ID: 723c0938-81d5-40af-b4b2-618efea7f6cb

rsawtell, 2020-10-06:
Also a problem with Run ID: b25a62fd-6976-49d3-ac5e-1bac576c1747

nkoenig, 2021-01-13:
Closing in order to triage issue tracker and due to the transition to Ignition Dome.

closed 2021-01-13 by nkoenig