osrf/subt #643: Comms neighbors function broken

rsawtell, 2020-10-02:
By breadcrumbs and/or teambase.

Without either of these, it works as you would expect, with the most recent update listed as the last time robots communicated. With breadcrumbs and teambase, some robots report rssi connection even outside of comms range, and all timestamps continue to update. Additionally the 'inf' rating appears frequently even when the robots are next to each other or far apart, I'm unsure if this is supposed to mean an infinitely good connection or an infinitely bad one, but either way it shows up when it shouldn't.

Example neighbor reporting by robot X2, which is so far down the tunnel it should not be able to communicate with anyone.
`[ INFO] [1601649575.230062737, 507.456000000]: Now: 507.456000, Robot X2 has Neighbor Found TEAMBASE at time 507.340000 with signal strength inf
[ INFO] [1601649575.230198501, 507.456000000]: Now: 507.456000, Robot X2 has Neighbor Found X1 at time 507.456000 with signal strength -58.928108
[ INFO] [1601649575.230366682, 507.456000000]: Now: 507.456000, Robot X2 has Neighbor Found X2 at time 507.340000 with signal strength inf
`

acschang, 2020-10-02:
> By breadcrumbs and/or teambase.
I'm not seeing any adverse behavior with RSSI when a teambase is not present.  There is always `inf` RSSI to/from the teambase regardless of whether a robot is in range of out of range.  There also seems to be incorrect RSSI values (i.e. -20 RSSI when out of communication range) for robot to robot communication.

> Without either of these, it works as you would expect, with the most recent update listed as the last time robots communicated. With breadcrumbs and teambase, some robots report rssi connection even [...]
I've seen RSSI ratings of `inf` when the RX/TX are next to one another and a packet is dropped but otherwise I see reasonable RX/TX values.

Based upon my experiments, it seems that the teambase is affecting RSSI.  This might be the root cause for the inaccuracy on the `Neighbors()` function call.

For these experiments I'm using the [setup outlined here](https://github.com/osrf/subt/tree/master/subt-communication).  On the left is the Ignition GUI with communication visualization with the X1 vehicle location coordinate XYZ coordinate frame shown whereas the bottom right shows the interface for `subt_comms_tester.py` then in the top right the left-most `rostopic echo` is the RSSI history for X1->X2 while the rightmost is the RSSI history for X1->TEAMBASE.  The X2 and TEAMBASE are both at their initial starting locations in the staging area.

**No Teambase:**
Before breadcrumb deployment:
[image]
[image]
[image]

After breadcrumb deployment:
[image]
[image]

**With a Teambase:**
Before breadcrumb deployment:
[image]
[image]

After breadcrumb deployment:
[image]
[image]
[image]
[image]

rsawtell, 2020-10-02:
In our case at least, the biggest problem is that it keeps updating the last contact timestamps when the robots are out of range in the presence of teambase/breadcrumbs (probably teambase based on what you are saying), as we were using them to determine when communications with a given robot have gone stale.

I just ran another check with no teambase, I think part of what I'm seeing is it almost seems like the reported RSSI are scrambled with one another:
`[ INFO] [1601664485.411766952, 30.448000000]: Now: 30.448000, Robot X1 has Neighbor Found X1 at time 30.404000 with signal strength -26.376148
[ INFO] [1601664485.411844736, 30.448000000]: Now: 30.448000, Robot X1 has Neighbor Found X2 at time 30.440000 with signal strength -25.464403
[ INFO] [1601664489.242419835, 30.948000000]: Now: 30.948000, Robot X1 has Neighbor Found X1 at time 30.900000 with signal strength -19.605533
[ INFO] [1601664489.242632026, 30.948000000]: Now: 30.948000, Robot X1 has Neighbor Found X2 at time 30.940000 with signal strength inf
...
[ INFO] [1601665111.151381610, 107.444000000]: Now: 107.444000, Robot X1 has Neighbor Found X1 at time 107.400000 with signal strength -52.569510
[ INFO] [1601665111.151440056, 107.444000000]: Now: 107.444000, Robot X1 has Neighbor Found X2 at time 107.416000 with signal strength inf
`

For X1 the RSSI for X1->X2 continues to drop while X2 is generally inf, with X1 staying in the starting area while X2 moves away from it.

X2 on the other hand reports:
`[ INFO] [1601665337.378846418, 135.548000000]: Now: 135.548000, Robot X2 has Neighbor Found X1 at time 135.500000 with signal strength -81.885303
[ INFO] [1601665337.378928218, 135.548000000]: Now: 135.548000, Robot X2 has Neighbor Found X2 at time 135.528000 with signal strength inf

`
which seems correct.

Additionally, the timestamps kept updating even after the robots lost contact with one another without any teambase/breadcrumbs present, so that part may not be teambase specific. Am I misunderstanding the timestamps for the neighbors function? I thought they indicated the last time a robot successfully communicated with another and had successfully tested some code based on that assumption a few days ago but it seems to not be working now as the timestamps always update even out of comms range.

peci1, 2020-10-03:
Just to check - do you run each robot with a different ROS_MASTER_URI? (i.e. via docker compose)

arjo129, 2020-10-13:
This is kind of a deal breaker for us cause we rely on RSSI to lay our mesh network :cry:

snkitche, 2021-07-07:
Any updates on this issue?